vlachoudis/bCNC #941: Tool Change not working

micksulley, 2018-08-19:
I suspect that I am doing something wrong here, but tried a few things and researched it and cannot find a solution.  
Running bCNC 0.9.14 on a Raspberry Pi.  g-code generated from FreeCAD.
I have gone through the WiKi and setup tool change, when I press the Change button it works as expected, moves to the change position, when I press Resume it moves to the probe position and runs a probe cycle.
However, it just ignores the tool change instructions in the code.
Here is my setup
[image]
and my g-code tool change is

(begin operation: T3: Engrave)
(Path: T3: Engrave)
(T3: Engrave)
(begin toolchange)
; M6 T3.0
M3 S0.0000
(finish operation: T3: Engrave)
(begin operation: Engrave)
(Path: Engrave)
(Engrave)

I don't know why M6 is preceded by a semicolon, does that matter?  When I look at the terminal screen as it runs the M6 line does not appear.

Any suggestions how I can get it to work?
Thanks
Mick

MARIOBASZ, 2018-08-19:
Semicolon indicates that it is a comment

MARIOBASZ, 2018-08-19:
grbl does not understand M06, so he puts a semicolon