vlachoudis/bCNC #942: I can not open last version and legend in previous version

MARIOBASZ, 2018-08-20:
I can not open the latest version and it works fine but this legend appears in previous version

[image]

"No module named enum "

Meaning? Does someone happen to you, mainly not being able to use the latest version? solution?

Harvie, 2018-08-21:
You have to install "enum" module to python:
https://pypi.org/project/enum/

Can be installed using following command `pip2.exe install enum`

It's obviously dependency of STL reader. I've missed this one. However there are several solutions:

  - include it to ./lib/ in bCNC, but that's not possible, since enum is GPLv3+ :-( , maybe we can ask the authors to dual-license
  - it will get fixed by itself once bCNC will run on Python 3, since enum module is included in Python 3 by default
   - STL parser is not needed for slicing, we can write smaller STL parser, i wanted to do that anyway, because current STL parser is more complex than needed, it's on my TODO list (but with very very low priority). Anyway it shouldn't be too hard to eliminate enum dependency altogether to get rid of this problem.

Harvie, 2018-08-21:
Ok, this is fixed now. But you still need to install numpy.

MARIOBASZ, 2018-08-21:
I have installed numpy, download from your branch, bcnc does not start

Harvie, 2018-08-21:
@MARIOBASZ what does it say?

MARIOBASZ, 2018-08-21:
numpy I have it installed before, I think you need it is necessary to use the camera
C: \ Users \ User \ Desktop \ Python27 \ Lib \ site-packages
I do not know why, but it does not start

Harvie, 2018-08-21:
@MARIOBASZ so it does still complain about numpy even when it's installed?

MARIOBASZ, 2018-08-21:
@Harvie, mi problema es que BCNC no se inicia, parece que inicia y cierra BCNC y la ventana de python

Harvie, 2018-08-21:
launch it from commandline, or othervise get the output of it.
eg.: put pause at the end of bCNC.bat

```
@echo off
set DIR=%~dp0
set PYTHONPATH=%DIR%lib;%DIR%plugins;%PYTHONPATH%
start python "%DIR%bCNC.py"
pause
```

or maybe

```
@echo off
set DIR=%~dp0
set PYTHONPATH=%DIR%lib;%DIR%plugins;%PYTHONPATH%
python "%DIR%bCNC.py"
pause
```

Harvie, 2018-08-21:
Try installing Pillow (also known as PIL) module

MARIOBASZ, 2018-08-22:
Here it is already late, tomorrow I will try, thank you very much!

MARIOBASZ, 2018-08-22:
Hello, I downloaded Pillow and the problem continues. From the command line I followed your suggestion, this legend appears:
image=Utils.icons ["new"]
KeyError: 'new'
[image]

I hope it is solved and download the version with your latest advances

alk1913, 2018-08-22:
helloi think that is something related with the icons extension.
try this
[bCNC-master (22).zip](https://github.com/vlachoudis/bCNC/files/2312341/bCNC-master.22.zip)

Harvie, 2018-08-23:
@alk1913 did you converted the file format, or just changed the extension?

alk1913, 2018-08-23:
Hello,i change the file format

Harvie, 2018-08-23:
I've converted images back to GIFs and commited the changes...

MARIOBASZ, 2018-08-23:
Thanks, it's already open

closed 2018-08-28 by MARIOBASZ