vlachoudis/bCNC #943: i can't start bcnc

darmalodar, 2018-08-21:
Microsoft Windows [Versión 6.1.7601]
Copyright (c) 2009 Microsoft Corporation. Reservados todos los derechos.

C:\Windows\system32>cd..

C:\Windows>cd..

C:\>cd bcnc

C:\bcnc>dir
 El volumen de la unidad C es 64Win7
 El número de serie del volumen es: B815-16CC

 Directorio de C:\bcnc

21/08/2018  01:26 a.m.    <DIR>          .
21/08/2018  01:26 a.m.    <DIR>          ..
13/08/2018  03:29 a.m.                28 .gitignore
13/08/2018  03:29 a.m.                 0 .git_cache_meta
13/08/2018  03:29 a.m.               224 bCNC
13/08/2018  03:29 a.m.               102 bCNC.bat
13/08/2018  03:29 a.m.               160 bCNC.desktop
13/08/2018  03:29 a.m.             1,150 bCNC.ico
13/08/2018  03:29 a.m.             5,686 bCNC.ini
13/08/2018  03:29 a.m.            19,656 bCNC.png
13/08/2018  03:29 a.m.            67,980 bCNC.pot
13/08/2018  03:29 a.m.            80,488 bCNC.py
13/08/2018  03:29 a.m.            33,236 bCNC.xbm
13/08/2018  03:29 a.m.             9,192 Camera.py
21/08/2018  12:20 a.m.             8,240 Camera.pyc
13/08/2018  03:29 a.m.           136,581 CNC.py
21/08/2018  12:20 a.m.           101,137 CNC.pyc
13/08/2018  03:29 a.m.            72,585 CNCCanvas.py
21/08/2018  12:20 a.m.            59,297 CNCCanvas.pyc
13/08/2018  03:29 a.m.            29,290 CNCList.py
21/08/2018  12:20 a.m.            25,126 CNCList.pyc
13/08/2018  03:29 a.m.             5,941 CNCRibbon.py
21/08/2018  12:20 a.m.             5,919 CNCRibbon.pyc
13/08/2018  03:29 a.m.            38,489 ControlPage.py
21/08/2018  12:20 a.m.            37,875 ControlPage.pyc
13/08/2018  03:29 a.m.            19,703 EditorPage.py
21/08/2018  12:20 a.m.            20,411 EditorPage.pyc
13/08/2018  03:29 a.m.            10,553 FilePage.py
21/08/2018  12:20 a.m.            10,465 FilePage.pyc
13/08/2018  03:29 a.m.    <DIR>          icons
21/08/2018  12:45 a.m.    <DIR>          lib
13/08/2018  03:29 a.m.             1,763 LICENSE.BSD3
13/08/2018  03:29 a.m.                13 LICENSE.md
13/08/2018  03:29 a.m.             1,291 LICENSE.MIT
13/08/2018  03:29 a.m.    <DIR>          locale
13/08/2018  03:29 a.m.               343 makefile
13/08/2018  03:29 a.m.    <DIR>          pendant
13/08/2018  03:29 a.m.             7,379 Pendant.py
21/08/2018  12:20 a.m.             7,609 Pendant.pyc
21/08/2018  12:45 a.m.    <DIR>          plugins
13/08/2018  03:29 a.m.            61,726 ProbePage.py
21/08/2018  12:20 a.m.            52,084 ProbePage.pyc
13/08/2018  03:29 a.m.             3,163 README.md
13/08/2018  03:29 a.m.            21,543 Ribbon.py
21/08/2018  12:20 a.m.            19,322 Ribbon.pyc
13/08/2018  03:29 a.m.            44,250 Sender.py
21/08/2018  12:20 a.m.            40,336 Sender.pyc
13/08/2018  03:29 a.m.             6,451 TerminalPage.py
21/08/2018  12:20 a.m.             6,505 TerminalPage.pyc
13/08/2018  03:29 a.m.            53,788 ToolsPage.py
21/08/2018  12:20 a.m.            50,572 ToolsPage.pyc
13/08/2018  03:29 a.m.             6,105 Updates.py
21/08/2018  12:20 a.m.             6,222 Updates.pyc
13/08/2018  03:29 a.m.            22,101 Utils.py
21/08/2018  12:20 a.m.            21,639 Utils.pyc
21/08/2018  12:29 a.m.    <DIR>          __pycache__
              48 archivos      1,233,719 bytes
               8 dirs  264,922,771,456 bytes libres

C:\bcnc>python bcnc.py
Traceback (most recent call last):
  File "C:\bcnc\ToolsPage.py", line 934, in __init__
    exec("import %s"%(name))
  File "<string>", line 1, in <module>
  File "C:\bcnc\plugins\slicemesh.py", line 26, in <module>
    import numpy as np
ImportError: No module named numpy
Traceback (most recent call last):
  File "bcnc.py", line 2592, in <module>
    application = Application(tk)
  File "bcnc.py", line 113, in __init__
    self.ribbon = Ribbon.TabRibbonFrame(self)
  File "C:\bcnc\Ribbon.py", line 494, in __init__
    image=Utils.icons["new"],
KeyError: 'new'

C:\bcnc>

Harvie, 2018-08-21:
Currently you need to install following python libraries to run development version:
numpy, scipy, enum

I am working on eliminating these dependecies, but i probably can't get away without numpy at all and without scipy some features would be slowed down too much.

Harvie, 2018-08-21:
Now you should only need numpy: https://pypi.org/project/numpy/

darmalodar, 2018-08-21:
07/10/2017  08:07 a.m.           228,621 LinuxCNC_Getting_Started.pdf
09/10/2017  01:44 a.m.    <DIR>          Mach3
25/11/2017  11:44 p.m.    <DIR>          miniBloq.v0.83
01/12/2006  11:37 p.m.           904,704 msdia80.dll
04/09/2017  03:56 p.m.    <DIR>          My Drivers
14/07/2009  12:20 a.m.    <DIR>          PerfLogs
21/09/2017  10:57 p.m.                 0 places.sqlite
08/10/2017  08:44 p.m.    <DIR>          Predator Software
21/08/2018  12:24 a.m.    <DIR>          Program Files
13/08/2018  03:11 p.m.    <DIR>          Program Files (x86)
21/08/2018  07:32 p.m.    <DIR>          pyserial-3.3
21/08/2018  12:29 a.m.    <DIR>          pyserial-3.4
21/08/2018  12:03 a.m.    <DIR>          Python27
05/05/2018  05:49 p.m.    <DIR>          ShanaEncoder
25/01/2018  10:07 p.m.    <DIR>          temp
21/08/2018  06:45 p.m.    <DIR>          totalcmd
20/05/2018  11:10 a.m.    <DIR>          Users
21/08/2018  12:27 a.m.    <DIR>          Windows
29/07/2018  10:32 a.m.    <DIR>          Windows_Repair_Toolbox
               8 archivos     13,296,423 bytes
              23 dirs  263,751,094,272 bytes libres

C:\>cd pyserial-3.3

C:\pyserial-3.3>dir
 El volumen de la unidad C es 64Win7
 El número de serie del volumen es: B815-16CC

 Directorio de C:\pyserial-3.3

21/08/2018  07:32 p.m.    <DIR>          .
21/08/2018  07:32 p.m.    <DIR>          ..
07/03/2017  07:32 p.m.            23,993 CHANGES.rst
07/03/2017  10:44 p.m.    <DIR>          documentation
21/08/2018  07:33 p.m.    <DIR>          enum-0.4.6.tar
21/08/2018  12:59 a.m.            20,265 enum-0.4.6.tar.gz
07/03/2017  10:44 p.m.    <DIR>          examples
02/05/2016  09:29 p.m.             1,885 LICENSE.txt
29/03/2016  04:31 p.m.             1,074 MANIFEST.in
21/08/2018  07:01 p.m.        13,470,769 numpy-1.15.1-cp27-none-win_amd64.whl
21/08/2018  07:01 p.m.        13,470,769 numpy.whl
07/03/2017  10:44 p.m.             1,612 PKG-INFO
18/02/2016  06:57 p.m.            12,497 pylintrc
07/03/2017  10:44 p.m.    <DIR>          pyserial.egg-info
20/12/2016  07:03 p.m.             1,763 README.rst
07/03/2017  10:44 p.m.    <DIR>          serial
07/03/2017  10:44 p.m.               146 setup.cfg
18/06/2016  04:41 p.m.             2,948 setup.py
07/03/2017  10:44 p.m.    <DIR>          test
              11 archivos     27,007,721 bytes
               8 dirs  263,751,094,272 bytes libres

C:\pyserial-3.3>cd enum-0.4.6.tar,gz
El sistema no puede encontrar la ruta especificada.

C:\pyserial-3.3>cd enum-0.4.6.tar.gz
El nombre del directorio no es válido.

C:\pyserial-3.3>cd enum-0.4.6.tar.gz
El nombre del directorio no es válido.

C:\pyserial-3.3>dir
 El volumen de la unidad C es 64Win7
 El número de serie del volumen es: B815-16CC

 Directorio de C:\pyserial-3.3

21/08/2018  07:37 p.m.    <DIR>          .
21/08/2018  07:37 p.m.    <DIR>          ..
07/03/2017  07:32 p.m.            23,993 CHANGES.rst
[... 205 lines cut ...]

C:\bcnc>python bcnc.py
Traceback (most recent call last):
  File "C:\bcnc\ToolsPage.py", line 934, in __init__
    exec("import %s"%(name))
  File "<string>", line 1, in <module>
  File "C:\bcnc\plugins\slicemesh.py", line 32, in <module>
    import stl #FIXME: write smaller STL parser
  File "C:\bcnc\lib\stl\__init__.py", line 1, in <module>
    from .stl import BUFFER_SIZE
  File "C:\bcnc\lib\stl\stl.py", line 12, in <module>
    from . import base
  File "C:\bcnc\lib\stl\base.py", line 21, in <module>
    class Dimension(enum.IntEnum):
AttributeError: 'module' object has no attribute 'IntEnum'
Traceback (most recent call last):
  File "bcnc.py", line 2592, in <module>
    application = Application(tk)
  File "bcnc.py", line 113, in __init__
    self.ribbon = Ribbon.TabRibbonFrame(self)
  File "C:\bcnc\Ribbon.py", line 494, in __init__
    image=Utils.icons["new"],
KeyError: 'new'

C:\bcnc>

darmalodar, 2018-08-21:
intalled enum y numpy and it shows me what i send

mexbravo, 2018-08-22:
same problem

alk1913, 2018-08-22:
try this. i only change the icons extension from GIF like in the last functional clone
[bCNC-master (22).zip](https://github.com/vlachoudis/bCNC/files/2312188/bCNC-master.22.zip)

mexbravo, 2018-08-22:
thanks...with this version work perfect

Harvie, 2018-08-23:
Yes, the problem is caused by me changing icons from GIF to PNG. It's strange that for me PNG works without problem. I think that it might be caused by the fact that i have Pillow module installed, which adds PNG support to Tkinter. I will revert PNGs back to GIF, but i'd like to understand this issue better, so please can you tell me if installing Pillow helps?

mexbravo, 2018-08-23:
I have Pillow installed too, and doesnt work with PNG

MARIOBASZ, 2018-08-23:
I also download Pillow in windows and it did not work. I will try to return from png to gif

Harvie, 2018-08-23:
I've converted images back to GIFs and commited the changes...

Laopa, 2018-08-24:
Hi

I got same issue, now with BCNC(22) it is nearly ok.  Mean BCNC can be open and launch, but if look at the CMD windows, find than missing some PYTHON module. nympy.....

To get ride of all anoying problems,  i add this command line in administrator CMD windows to install all necessary modules in PYTHON2.7.13.

"python -m pip install --user numpy scipy matplotlib ipython jupyter pandas sympy nose"

ALSO one very important things and not mention anywhere is YOU MUST INSTALL PYTHON2.7 X32. the X64 does not work. Also PYSERIAL is only for X32.

Happy CNC.

Harvie, 2018-08-24:
> "python -m pip install --user numpy scipy matplotlib ipython jupyter pandas sympy nose"

AFAIK you don't need all of these, probably only numpy.

cz7asm, 2018-08-29:
I had the same problem running it on Windows:

```
Traceback (most recent call last):
  File "bCNC.py", line 2592, in <module>
    application = Application(tk)
  File "bCNC.py", line 113, in __init__
    self.ribbon = Ribbon.TabRibbonFrame(self)
  File "D:\devel\mech\bCNC-master\Ribbon.py", line 495, in __init__
    image=Utils.icons["new"],
KeyError: 'new'
```

It turned out it was caused by this in Utils.py:

```
               try:
			icons[name] = PhotoImage(file=img)
			if getBool("CNC", "doublesizeicon"):
				icons[name] = icons[name].zoom(2,2)
		except TclError:
			pass
```
tkinter doesn't work with png images (at least on my win version) and the exception doesn't end up being very informative!

**The solution** (info here - http://effbot.org/tkinterbook/photoimage.htm):

```
                from PIL import Image, ImageTk
           ...
		try:
                        image = Image.open(img)
			icons[name] = ImageTk.PhotoImage(image)
			if getBool("CNC", "doublesizeicon"):
				icons[name] = icons[name].zoom(2,2)
		except TclError as e:
                        print e
			pass
```

Harvie, 2018-08-29:
@cz7asm if you read this issue closely, you will find that i've removed the need for pillow and that it waits for upstream merge. you can try new version which does not need pillow from my fork:
http://github.com/harvie/bCNC/

cz7asm, 2018-08-29:
@Harvie right, I gave it as much time as this issue deserved :)

Guenni75, 2018-09-07:
I had the same Problem and the Harvie-Version works.

closed 2018-10-30 by Harvie