vlachoudis/bCNC #945: Tool Change Information

micksulley, 2018-08-21:
This is not an issue as such but I didn't know how to post information for bCNC
I have been testing the Tool Change function and am really impressed, but I did find the documentation a bit sketchy, so I created a document for it and thought someone else may find it useful.

[ToolChangeV01.doc.zip](https://github.com/vlachoudis/bCNC/files/2307946/ToolChangeV01.doc.zip)

Harvie, 2018-08-22:
I think this should go to wiki: https://github.com/vlachoudis/bCNC/wiki
I am not sure if you have permissions to edit it, but i suppose you do.

micksulley, 2018-08-22:
Just tried it and it seems I do have permission, it is there.

MARIOBASZ, 2018-08-23:
It is a good help to get involved in editing guides!
I have not seen your contribution uploaded to the page yet, I hope they upload it soon

Harvie, 2018-08-23:
It's already there: https://github.com/vlachoudis/bCNC/wiki/Tool-Change-with-bCNC,-FreeCAD-&-grbl

closed 2018-10-30 by Harvie