vlachoudis/bCNC #947: Probe Feed Rate not working

micksulley, 2018-08-24:
Running version 0.9.14 on Raspberry Pi.
Just looking at the Probe function and trying out options and changing the Fast Probe Feed and/or Probe Feed does not make any difference.  I am testing by probing 1mm, takes about 8 seconds to do it but I have tried changing both feed rates by a factor of 10 up and down, and it still takes about 8 seconds.  Have I missed something here?
Also I saw mention in the issues of a fast probe, then pull off and slow probe to get accuracy.  Is there some setup to enable that?
Thanks
Mick

MARIOBASZ, 2018-08-25:
Is it weird, do you have low acceleration? Trying with a bigger distance?

micksulley, 2018-08-25:
I just tested it again, bigger distance still gave the same result, however it seems there is a minimum rate for Probe Feed.  I had it set to 10 and tried reducing it to 1 or 0.1, no difference.  I put it up to 100 and yes it is faster.  Quick trial & error suggests that somewhere around 20 is the lowest it will go, below that the rate does not change.
Can you tell me what the Fast Probe Feed should do?  There doesn't seem to be anything in the documentation for that.

MARIOBASZ, 2018-08-25:
Did you notice the help?
https://github.com/vlachoudis/bCNC/wiki/AutoLevel
Fast Probe Feed is the speed from one point to the next point in the xy plane
Probe Feed is the drilling speed on the Z axis. Was this what you asked?
The problem could be the values ​​of speed, fast speed and acceleration established in GRBL?

micksulley, 2018-08-25:
OK I think I have got to the bottom of this.  I tested over a 5mm probe distance
Feed Rate    Time
20                 15
10                 30
9                   34
5                   34
0.1                34
It turns out that there is a minimum speed that gRBL can handle, send it anything lower and it goes at its minimum, so it seems that the min Z for my machine is about 0.15 mm/second.  A bit unexpected but now that I know I can live with it.
Thanks for your help

closed 2018-08-25 by micksulley

MARIOBASZ, 2018-08-26:
ahhh ok, 9mm / min would be, I did not know that data, always use much higher values.