vlachoudis/bCNC #948: Impossible to install bCNC on MAC

lbaumstark, 2018-08-25:
Hello, 
I've got a lot of problems to launch and use bCNC. 
I installed python2, 
I work on Mac 10.13.2
When I run BCNC.py I've got this error : 

Traceback (most recent call last):
  File "/Users/luc/Downloads/bCNC-master/ToolsPage.py", line 934, in __init__
    exec("import %s"%(name))
  File "<string>", line 1, in <module>
  File "/Users/luc/Downloads/bCNC-master/plugins/slicemesh.py", line 26, in <module>
    import numpy as np
ImportError: No module named numpy 

Thanks a lot for your help.
Luc

Harvie, 2018-08-25:
Install numpy module to python

toktuff, 2018-08-25:
I think we should keep new dependencies to a minimum. I.e. the mesh slicer is something that perhaps 5% of bCNC users will use and cause headaches for 100% of them.

Is there a way to delay load of the plugins so that these errors are triggered only if the plugin is used?

closed 2018-10-30 by Harvie