vlachoudis/bCNC #954: Autoleveling extrapolation outside of probed margins goes crazy

Harvie, 2018-08-29:
When some g-code gets outside of the autolevel margins, the height ~~interpolation~~ extrapolation goes all crazy on us. Here is example of toolpath much bigger than probe data:

[image]

Selected paths are approximately the probed area. Here on XZ plot you can see, how extrapolation goes completely nuts outside of that area:

[image]

cliff-pennalligen, 2021-01-09:
Hello, just wondering whether anything was actually done with this?  I thought of a work-around meanwhile - save the probe file and extrapolate manually, reload and cut.

MARIOBASZ, 2021-01-09:
That is one of the problems. The other one is also dangerous: if there is also a G53 (I use it at the end to go to machine safe Z: G53 G0 Z-5.00), treat it as G0 Z-5.00 in the working area (G54 or whatever) . I have ruined equipment and tools like that.

cliff-pennalligen, 2021-01-10:
> <comment@noreply.github.com>
> of probed margins goes crazy (#954)

Fortunately my router is not powerful enough to easily wreck itself... 
although the issue in my case usually lies between the keyboard and the 
back of the chair!

@vlachoudis/bCNC, an idea - as an interim measure, would it be hard to 
scan for currently unsupported commands and warn the user prior to using 
auto level?

Cliff Pennalligen Project Engineer
CNC Design Ltd
Certified Siemens Solution Partner
ph: +64 (0)9 274 1280 mob: +64 (0)275 740 124 email: 
cliff.pennalligen@cncdesign.co.nz office: 9A Newark Place, East Tamaki, 
Auckland www.cncdesign.co.nz