vlachoudis/bCNC #955: Lathe Orientation/Visualizer and 2 axis only

MadTooler, 2018-08-31:
Thanks for the excellent bit of software!  My typical applications are industrial, but I am having fun so far with the extensive features packed into bCNC.

I am currently trying to test an old CNC wood lathe with bCNC, rpi3 with 7" touchscreen and kvkbd on screen keyboard, and GRBL 1.1.  I believe I have all of the current versions installed as of today.

Is there a way to change the layout from a 3 axis mill to a 2 axis lathe?

Ideally, I am looking for both a GUI with only the 2 axis shown and the "canvas" rotated with the Z horizontal.

Thanks

tklus, 2018-09-06:
That would be an awesome feature! Maybe a button we could push to switch between lathe and mill coordinate systems.

I am currently converting a manual lathe to cnc and this would be a big help!

JWood1w4, 2019-04-25:
Any movement on this?

plankatron, 2019-10-25:
Bump for visibility, I would even go so far as to figure out some kind of feature bounty for this feature as it is something that I bet quite a few people would be interested in!

Harvie, 2019-12-01:
This non-3-axes stuff is requested a lot. I don't have such machine, but maybe there is enough people to establish some team for this:
https://github.com/vlachoudis/bCNC/issues?utf8=%E2%9C%93&q=label%3A%22more+than+3+axes%22+

My only concern is that bCNC does not get cluttered by various groups of people hacking together some support for their axis configuration. We need to address this in some modular manner, so everyone's happy, but it does not turn into the cluttered UI and spaghetti code.