vlachoudis/bCNC #956: Autoleveling/Digitizing does not need to go all the way to the safe Z every time

Harvie, 2018-09-01:
It's possible to speed the 3D probing by big time, if you don't go to safe Z between measuring points. However you still need to move using probe, so you don't crash.

https://www.youtube.com/watch?v=LvSeUWyTnCU

walterwissmann, 2018-09-01:
Yes it is wery nice! And nicer with this:
https://www.youtube.com/watch?v=4a44KysPykg

Harvie, 2018-09-01:
@walterwissmann impressive build for DIY project. However bCNC should work with whatever hardware you have.

walterwissmann, 2018-09-02:
Yes.
The mosts have a digitizer under the spindelmotor.
But my thoughts are around the rotating mechanics as in the example video of me...

bCNC should have a function to take into account that the digitizer can be mounted to another position.
Similar to the 'Probe Camera Alignment'...

Here a cheap digitizer for testing (Sorry in German, but scroll down and see the foto's)

walterwissmann, 2018-09-02:
@MARIOBASZ: 
For me it is not a problem to design a optical Sensor for scanning...
Can you generate the program code for scanning?

Harvie, 2018-09-02:
@MARIOBASZ There is no need for aditional sensor, because you already have probe, which can be used even in XY movements. This can be done purely in software (see the first video, they only use regular probe).
@walterwissmann Having two sensors is overcomplicated, but it would be very cool if you could use optical sensor ONLY instead of touch probe. But i've been doing some tests and i didn't manage to come up with something precise enough... It's hard to focus optical sensors to tiny spot (eg. 0.1mm) in way that it would not be affected by ambient light, reflectiveness of probed part and surrounding shapes.

walterwissmann, 2018-09-05:
Laser DistanceSensor, TypicalResolution 0,04mm :
https://www.sensopart.com/en/products/distance-sensors/distance-sensors/ft-50-rla-20-f-l4s.html

Harvie, 2018-09-05:
@walterwissmann i've been looking at these sensors. correct me if i am wrong, but i think that the 0.04 resolution applies when the sensor is moving towards the flat surface with constant reflectivity (eg. distance from sensor to the wall). you can't use it to scan shapes with complex details (eg. coins), because the cone of light used to measure is not focused to single tiny spot.

Harvie, 2018-09-05:
This is more hardware issue than bCNC's. But when we're at it, i've just filled issue at grbl, which might enable more complex sensor setups: https://github.com/grbl/grbl/issues/1433

walterwissmann, 2018-09-05:
Yes, i am thinking also...
The DataSheet sayed: **Size of light spot : < 1 mm at 50 mm**
For a PCB it should be enough... 
The Price: 399,-€ :-(

MARIOBASZ, 2018-09-05:
I thought of something like CNY70. Maybe it's better VL5310 or VL6180

Harvie, 2018-09-05:
@MARIOBASZ i've been already looking on these exactly. also i was thinking about attaching optical fiber to CNY70, so i can direct the light to the tiny tip. I've been trying this approach, i've cleaved fiber and i've been shining the laser into it and watching for reflection that goes back to the fiber from probed surface, but i've found that it's not focused enough to be precise. Maybe some lenses might make this work. dunno.

walterwissmann, 2018-09-05:
Hmm,
@MARIOBASZ and @Harvie 
should we discuss this on an an other Place...?

MARIOBASZ, 2018-09-07:
@ HARVIE, witty thought, I am reasoning by another way.