vlachoudis/bCNC #958: BCNC Does Not Connect Shield - Idle Terminal Message

cncvictor, 2018-09-02:
Hello guys

My BCNC has stopped connecting with the shield, I click to connect and it does not connect and also in the terminal of the software is a message of idle, idle, idle, it goes down without stopping.

Can someone help me please?

Victor

cncvictor, 2018-09-02:
[image]
[image]
I click open and the message is left unconnected. I do the $$ command and the <Idle> message goes on and on. All the Arduino lights are on, it's working, but there's something wrong that I still can not identify.
Please, if anyone can help me, I thank you in advance.

cncvictor, 2018-09-02:
[image]
By no means does he make the connection

Victor

toktuff, 2018-09-02:
You can try to connect with some other program to try to narrow down the problem. I’ve used minicom in the past.

cncvictor, 2018-09-02:
[image]
[image]
[image]
I changed the program, now I tried to run the Universal Gcode Sender 2.0, but the program reports this error according to the images. I also tried Universal Gcode Sender 1.09, it opened, but did not obey the commands as well. What's going on? It's hard to solve this.

walterwissmann, 2018-09-02:
Hmm, it seems to me, that anything else with the UNO is wrong.
Time to Update gbrl to the newest version;-)

cncvictor, 2018-09-02:
I have a mini CNC, those with pen, I'm studying the whole mechanism to be able to make a bigger one. Take a look on my youtube channel that you will see my machine when it was working up to 1 week ago

look https://youtu.be/xUxPWlC78ZE

walterwissmann, 2018-09-02:
Nice,  Victor....
have you make an update of grbl?

cncvictor, 2018-09-02:
Yes, I did

walterwissmann, 2018-09-02:
And?

Harvie, 2018-10-30:
Have you resolved this problem?