vlachoudis/bCNC #959: Starting BCNC

Guenni75, 2018-09-02:
Hi. I had to setup a new Win10 PC.
I installed Python 2.7 and Numpy and Tkinter and so on but every time i get these error starting bcnc.py.

Traceback (most recent call last):
  File "bcnc.py", line 2592, in <module>
    application = Application(tk)
  File "bcnc.py", line 113, in __init__
    self.ribbon = Ribbon.TabRibbonFrame(self)
  File "D:\bCNC-master\Ribbon.py", line 494, in __init__
    image=Utils.icons["new"],
KeyError: 'new'

Every time i setup a PC (Win or Linux) with Python for Bcnc, i have other Errors and Problems because the Wiki is, in this Point, not completed.

Any one an idea what's the problem?

walterwissmann, 2018-09-02:
Hi Guenni,
have you read this thread:
https://github.com/vlachoudis/bCNC/issues/943

Guenni75, 2018-09-02:
Hi.
The Version from harvie works!!

Thanks.

walterwissmann, 2018-09-02:
:-)

Guenni75, 2018-09-09:
Is there still the Problem with the PNGs at vlachoudis Version?
And is the Harvie Version the same as the vlachous Version?

Harvie, 2018-10-30:
use latest master and install numpy

closed 2018-10-30 by Harvie