vlachoudis/bCNC #960: To open all code on the screen

sfinexer, 2018-09-06:
To open all code on the screen
It is necessary to remove restriction on quantity of the displayed code (blocks) on the screen, it is necessary for allocation of a part of a code and its editing (removal of the selected fragments).

In this case not all G-code which is in the file in case of his big size is displayed.

sfinexer, 2018-09-07:
Help to remove this restriction.
It prevents editing, in particular, removal of G-Code blocks because because of the fact that the point isn't displayed, can't allocate it.

Harvie, 2018-09-10:
Please increase this timeout value in corner of view:

[image]

This can't get properly fixed until #591 is solved.

sfinexer, 2018-09-10:
@Harvie.
It's a go.
In that case, it is necessary to display probably the message that loading is interrupted on the expiration of time.

Harvie, 2018-09-10:
Yes. Maybe. But i personaly think it would make more sense to invest time to bring real OpenGL support, which would provide faster graphics #591

Guenni75, 2018-09-12:
Real 3D in bcnc .... goose bumps!!

Harvie, 2018-10-30:
Closing duplicate of #591

closed 2018-10-30 by Harvie