vlachoudis/bCNC #964: Proper python configuration, using setuptools

petmakris, 2018-09-12:
I think this project needs proper configuration, 

a least a `setup.py` with reference to all project dependencies.

Should I go for it? Do you accept PR?

Thanks

Guenni75, 2018-10-13:
VERY GOOD IDEA!!!
Everytime i have Problems get BCNC work properly.
At the moment I have problems with the camera again!

Harvie, 2018-10-30:
@petmakris is that standard way of automaticaly installing dependencies in python?
if so, i will happily accept such PR.

Harvie, 2018-11-13:
@petmakris Can you please create the file? I am interrested to have this in next release.

petmakris, 2018-11-18:
Looking into it.

petmakris, 2018-11-18:
To structure a python properly means that you configure setuptools for your project so that you declare

- dependencies
- entrypoints (executable scripts)
- tests
- metadata 

Doing so allows for distributing your software as a (usually) a wheel and installing with a simple command

```
pip install bCNC
```

This command will fetch the bCNC wheel, will install bCNC's dependencies and will make available system wide the bCNC executable.

The structure of a `setup.py` file looks like
```python
from setuptools import setup, find_packages

setup(
    name = "bCNC",
    version = "0.1",
    packages = find_packages(),
    author = "Author",
    install_requires = [
        'cherrypy==18.0.1',
        'jinja2==2.10',
        'requests==2.20.1',
        'mysql-connector-python==8.0.13',
        'simplejson==3.16.0',

        # https://google-auth.readthedocs.io/en/latest/index.html
        'google-auth==1.6.1'
    ],

    entry_points = {
        'console_scripts': [
            'bCNC = {package}.{module}:{main_function}',

        ]
    }
)
```

and is documented here

https://setuptools.readthedocs.io/en/latest/setuptools.html

and I've written a tutorial on structuring a python project some time ago here

https://petmakris.github.io/structuring-a-python-project

While totally feasible and we should go for it the bCNC project is not structured at moment very well in terms of using python packages and imports and this will take some time. 

Let's not include it in the next release but I will work on it.

Harvie, 2018-11-19:
Wow! I really want to do this before next release!

Also this looks as good source: https://packaging.python.org/tutorials/packaging-projects/

Harvie, 2018-11-19:
OK. I've just started working on it and already run into problems:
`tkinter` and `python-imaging-tk` are not available as pip packages, therefore can't be installed automaticaly!

Harvie, 2018-11-19:
> While totally feasible and we should go for it the bCNC project is not structured at moment very well in terms of using python packages and imports and this will take some time.

Is it enough to move everything to `bCNC` subdirectory leaving only setup.py, LICENSE and README.md outside?

petmakris, 2018-11-19:
1) On tkinter & python-imagin-tk I have no opinion I have to check what's going on

2) On packaging, yes bCNC would be perfect as a name. In every directory inside bCNC/ and every subdirectory you need to add an __init__.py (this what makes the `folder` a `python module` that other .py can import from, and find_packages() can find it.)

petmakris, 2018-11-19:
`tkinter` is not a project dependency, it's included with python.

But for `python-imaging-tk` I don't what this is, it seems that it's part of PIL (python imaging library) or the same and seems to be replaced by Pillow (https://pillow.readthedocs.io/en/5.3.x/).

Maybe it's the same API and a seamless transition

petmakris, 2018-11-19:
Could you check first if the python dep you need is PIL ?

```
PIL (1.1.6)                 - Python Imaging Library
```

How did you install python-imaging-tk in the first place?

petmakris, 2018-11-19:
To give a hint

Do the following to check that your dependencies are working for your project.

In the bCNC root folder create a virtual environment

```python
python -m venv venv # python 3
```
or 
```python
pip install virtualenv # python 2
virtualenv venv
```

Source the virtualenv

```
source venv/bin/activate # linux
./venv/Scripts/activate # windows
```

Now if you do a `pip freeze` you'll see that there are no other deps that pip and setuptools (and maybe wheel)

Now in a clean virtualenv you do 

```
pip install -e .
```
(mind the dot after `-e`). This way you install all dependencies in the virtualenv and the bCNC for development (every change to bCNC files is available immediately in this `venv`)

This way you can test if the configured deps in `setup.py` are sufficient to start bCNC

Harvie, 2018-11-19:
> Could you check first if the python dep you need is PIL ?

Yes. It is. To be precise i use Pillow, which is PIL fork. But i think they are interchangeable.

I've set up virtual env and did pip2 install of following packages:

```
        install_requires = [
                'pyserial==3.4',
                'numpy==1.15.4',
                'opencv-python==3.4.2.17',
                'Pillow==5.3.0',
        ],
```

bCNC now works in this virtualenv. Does it mean that tkinter is included in python by default?

Harvie, 2018-11-19:
I have first build on PyPI, but it cannot import other files:

pip2 install bCNC --upgrade
python2 -m bCNC

```
Traceback (most recent call last):
  File "/usr/lib64/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib64/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/home/harvie/Temp/tst/venv/lib/python2.7/site-packages/bCNC/__main__.py", line 45, in <module>
    import Utils
  File "/home/harvie/Temp/tst/venv/lib/python2.7/site-packages/bCNC/Utils.py", line 13, in <module>
    from log import say
ImportError: cannot import name say
```

Harvie, 2018-11-19:
I've prepared the directory structure, but i still have troubles importing stuff. Also i am bit worried about how things like plugins will work out in this "python module" approach. Because loading plugins is based on enumerating directory and importing all the files that are found there. Not sure if that can be done in python module.

Harvie, 2018-11-19:
I needed to add some files to MANIFES.in, now everything installs and launches using pip. However it still gives me a ton of errors.

Harvie, 2018-11-19:
@petmakris Can you please try the module version? I need some people to test the module version of bCNC.

When i run bCNC as python module, none of plugins can save/load it's configuration. When i run bCNC directly it works. No error shown. How is that possible? Can you please take a look at this?

Harvie, 2018-11-19:
BTW i use ArchLinux for development, but i've just tried at Ubuntu 18.04.1 LTS and bCNC installed using pip2 install without any issues. OTOH on Ubuntu 14.04.5 LTS it didn't installed due to some problems with OpenCV or something... But i still think it's pretty cool to be able to install to latest Ubuntu using single command.

Harvie, 2018-11-20:
I think i am onto something. I've added following code to `Utils.py`:

```python
config    = ConfigParser.ConfigParser()
print "new-config", __name__, config 
```

When i run the bCNC as module, it creates two instances of config parser:

```
[harvie@anemophobia bCNC]$ python2 -m bCNC
new-config bCNC.Utils <ConfigParser.ConfigParser instance at 0x7fcca1177908>
new-config Utils <ConfigParser.ConfigParser instance at 0x7fcca0f94368>
```

When i run it directly it only creates single instance of that config parser:

```
[harvie@anemophobia bCNC]$ ./bCNC/bCNC
new-config Utils <ConfigParser.ConfigParser instance at 0x7faf2b9447e8>
```

That's because of multiple imports under different `__name__`

petmakris, 2018-11-20:
Sincere^2 apologies, I won't be able to check this out very soon, hard work :(

Everything you do though is in the right direction. This way of structuing python is the *only* way to go (and not just the recommended way) and it will reveal problems that would cause problem anyway.

Harvie, 2018-11-20:
Setup tools are implemented. Config is separate problem. I will close this and track it in #1053

closed 2018-11-20 by Harvie

Harvie, 2018-11-20:
Windows XP:

[image]