vlachoudis/bCNC #965: Strange results when trying to pocket 1 rectangle

DaveDavenport, 2018-09-12:
This is with the latest git (updated 12-09-2018, 8137b5082890572c657c9f76180732b88e9ea417).
On ubuntu 18.04 64bit.

If I import the following dxf:

[plank-display.zip](https://github.com/vlachoudis/bCNC/files/2375225/plank-display.zip)

Try to pocket the three rectangles in it, the first 2 go fine, the 3rd it goes wrong, the object becomes incredible large and outside of the original drawing:

Before pocket:
[image]

After

[image]

The dxf was created using librecad and are just rectangles.

DaveDavenport, 2018-09-12:
Tried the 0.9.11 tag, here it throws a big stack-trace:

https://gist.github.com/DaveDavenport/408cc3a348fbab6322f57857cdaaa7bc

walterwissmann, 2018-09-13:
Hello Dave,
i have testet it with @Harvie 's version
https://github.com/Harvie/bCNC

Here the Result:
[image]

Walter

Harvie, 2018-09-13:
I've also tried on my branch and got this:

```
Traceback (most recent call last):
  File "/home/harvie/Temp/bCNC/Utils.py", line 434, in __call__
    return self.func(*args)
  File "/home/harvie/Temp/bCNC/ToolsPage.py", line 1671, in execute
    self.tools.getActive().execute(self.app)
  File "/home/harvie/Temp/bCNC/ToolsPage.py", line 865, in execute
    app.pocket(name)
  File "bCNC.py", line 1781, in pocket
    msg = self.gcode.pocket(blocks, diameter, stepover, name)
  File "/home/harvie/Temp/bCNC/CNC.py", line 3921, in pocket
    newpath.extend(self._pocket(path, -D*diameter, stepover, 0))
TypeError: 'NoneType' object is not iterable
```

Harvie, 2018-09-13:
Problem is, that two rectangles on the sides are only 2.6mm wide, so they can't accomodate 3.2mm endmill! When i set 1mm endmill, everything worked OK. also it worked for @walterwissmann who had used 2mm endmill.

However there's bug which causes whole operation to fail, rather than ignore features that are too small and profiling/pocketing others. I am going to rewrite profile/pocket code from the scratch. So i will definitely fix this during that process.

DaveDavenport, 2018-09-14:
I was using a `2mm` endmill. (sorry forgot to mention).  i will try the harvie version.