vlachoudis/bCNC #967: 'probing' a drawing

swarfer, 2018-09-19:
Is there a way to use bCNC to digitize a drawing?

So if I put the drawing on my CNC bed and jog the tool/camera to relevant points in the drawing can I save all the points to a file (SVG/DXF/JSON/plain text) that I can then import into CAD.    
Then in CAD I could 'join the dots' to complete the drawing.

Any ideas?

swarfer, 2018-09-19:
I think I have it.
The probe page has the 'record' section which has RAPID  and FEED buttons.
These store the current tool position into the editor as Gcode.

Snag is that you have to swap panels between control and probe for each point which will make doing a drawing very tedious.
but, I think these lines
[code]
elif rexx.abbrev("RRAPID",cmd,6):
	Page.frames["Probe:Probe"].recordRapid()
elif rexx.abbrev("RFEED",cmd,5):
	Page.frames["Probe:Probe"].recordFeed()
[/code]
added to bCNC.py just after the 'ABOUT' section in execute() will allow shortcuts to 'rrapid' and 'rfeed'
making digitizing a hand drawing much much easier.

Next step is to track down a plugin for Sketchup to import the Gcode, or write one.

BTW I could not get the current master code to run and instead have loaded Harvie's fork.
Would much rather be using the real deal though.

walterwissmann, 2018-09-19:
Yes we are waiting to @vlachoudis to merge it. But he is still quiet now...

Harvie does a great job now...
Thanks to Harvie :-)

swarfer, 2018-09-20:
Refer to https://github.com/Harvie/bCNC/commit/bb1b614a50288d871914c5c7b8ed48af759e42c1
because yesterday I did not realize that the 'record' panel is only in Harvie's fork.

@Harvie maybe you would like to merge my 4 lines of code into your code?   I use shortcuts F7 and F8 for RRAPID and RFEED and it makes digitizing a breeze.

Will try a webcam tonight....

Harvie, 2018-09-20:
I've added following commands:

```
                # RR*APID:
                elif rexx.abbrev("RRAPID",cmd,2):
                        Page.frames["Probe:Probe"].recordRapid()

                # RF*EED:
                elif rexx.abbrev("RFEED",cmd,2):
                        Page.frames["Probe:Probe"].recordFeed() 

                # RP*OINT:
                elif rexx.abbrev("RPOINT",cmd,2):
                        Page.frames["Probe:Probe"].recordPoint()

                # RC*IRCLE:
                elif rexx.abbrev("RCIRCLE",cmd,2):
                        Page.frames["Probe:Probe"].recordCircle()

                # RFI*NISH:
                elif rexx.abbrev("RFINISH",cmd,3):
                        Page.frames["Probe:Probe"].recordFinishAll()

```

swarfer, 2018-09-20:
excellent, thankyou

closed 2018-10-30 by Harvie