vlachoudis/bCNC #968: bCNC randomly halts and terminates program mid-cycle

classbproject, 2018-09-19:
I've experienced this on two jobs now. The program just halts mid-cycle and terminates everything. Clicking on pause or stop has no effect. Here is an error message that it shows me. Can anyone help me understand why this happens and also how to fix this? 

	Program     : bCNC
	Version     : 0.9.14
	Last Change : 5 Feb 2018
	Platform    : win32
	Python      : 2.7.15 (v2.7.15:ca079a3ea3, Apr 30 2018, 16:30:26) [MSC v.1500 64 bit (AMD64)]
	TkVersion   : 8.5
	TclVersion  : 8.5

	Traceback:
	Traceback (most recent call last):
	  File "C:\nextcloud\electronics\Software\bCNC\Utils.py", line 434, in __call__
		return self.func(*args)
	  File "C:\nextcloud\electronics\Software\bCNC\Sender.py", line 753, in pause
		self.feedHold()
	  File "C:\nextcloud\electronics\Software\bCNC\Sender.py", line 733, in feedHold
		self.serial.write(b"!")
	  File "C:\Python27\lib\site-packages\serial\serialwin32.py", line 315, in write
		raise SerialException("WriteFile failed ({!r})".format(ctypes.WinError()))
	SerialException: WriteFile failed (WindowsError(22, 'The device does not recognize the command.'))

	Traceback (most recent call last):
	  File "C:\nextcloud\electronics\Software\bCNC\Utils.py", line 434, in __call__
		return self.func(*args)
	  File "C:\nextcloud\electronics\Software\bCNC\Sender.py", line 839, in stopRun
		self.feedHold()
	  File "C:\nextcloud\electronics\Software\bCNC\Sender.py", line 733, in feedHold
		self.serial.write(b"!")
	  File "C:\Python27\lib\site-packages\serial\serialwin32.py", line 315, in write
		raise SerialException("WriteFile failed ({!r})".format(ctypes.WinError()))
	SerialException: WriteFile failed (WindowsError(22, 'The device does not recognize the command.'))

Harvie, 2018-09-19:
Can you send us the g-code you are trying to run?
Do you have laster version of GRBL installed?

classbproject, 2018-09-19:
Yes grbl version 1.1f. The gcode is available here: [https://pastebin.com/cyYP13HA](https://pastebin.com/cyYP13HA).

Harvie, 2018-09-19:
I'll look at it later, but now i can tell you that for some reason i am unable to load your g-code in camotics g-code simulator... Maybe try different postprocessor in your cam?

walterwissmann, 2018-09-19:
It seems to me,
that is a problem with the CP2104 USB to UART bridge...

classbproject, 2018-09-19:
I read other issues here about this happening with FT232 and CH340G devices. Now the CP2104 also seems to be added to this list. These are very widely used USB-UART bridges and if this error occurs across all of them, then perhaps there is something wrong with the parser and not with the actual hardware. BTW, I ran four of these contour gcodes with [OpenCNCPilot](https://github.com/martin2250/OpenCNCPilot) and they all completed without any glitches.

Harvie, 2018-09-19:
For me it seem to run OK. Is it possible that you have some hardware problem? USB cable too long or too loose? or any other kind of loose connection? unshielded motor cables? fried arduino?

walterwissmann, 2018-09-19:
Ok...
OpenCNCPilot was created in C#. bCNC in Python (interpreter)... It can be, there are differend timings in the generated Code. On my RasPi 3+b bcnc has no Problem.

Harvie, 2018-09-19:
There's no legitimate reason why some brands of USB-Serial converters should corrupt g-code. If they do, they are buggy. I would suggest you to set your arduino (and bCNC) to lower baudrate. Dunno how this is done. I guess in case of GRBL you have to recompile. Not sure.

Harvie, 2018-09-19:
Yes. It's possible, that there's some "buffer overrun" in the usb-serial chip, which happens when data is transmitted too fast. This happend to me with cheap USB-MIDI converter. That's why i suggested lowering baudrate. Maybe it will even need to add some delay between g-code lines. But that's HW bug.

classbproject, 2018-09-19:
Did you run the code on a simulator or on an actual machine? If I remember reading correctly you cannot lower the baudrate with grbl 1.1. It has to be 115,200 or higher. 

The hardware is just fine since the jobs ran fine with other software. The end stops are noise filtered, the wires are long but shielded.

Harvie, 2018-09-19:
You can. But you have to recompile:
`#define BAUD_RATE 115200` in `config.h`

If you think that bCNC does not send the g-code correctly, you can use `interceptty` on Linux (or similar software on other platforms) to eavesdrop this communication as described in README.md, but i am not sure that bCNC would modify g-code before sending (except for toolchange macros if configured).

walterwissmann, 2018-09-19:
But: why can it be, that bCNC runns on a raspi3+b nice? ;-)

I found: Windows and his device drivers hase inner problems with fast hardware timings.
Also the electrical drifts with USB-Serial-Converters are make problems...
An example: https://stackoverflow.com/questions/34694052/why-serial-communications-from-arduino-stops-after-some-time

walterwissmann, 2018-09-19:
I am let my machine do the job now...
[image]

walterwissmann, 2018-09-19:
The first job is done:
[image]

walterwissmann, 2018-09-19:
Second Job:
[image]
It goes faster. I take the feed to 200%

walterwissmann, 2018-09-19:
Now the third job:
[image]

walterwissmann, 2018-09-19:
...
[image]

walterwissmann, 2018-09-19:
The end of the forth Job:
[image]

walterwissmann, 2018-09-19:
I can repeat this more and more...
bCNC workes great!

@classbproject : trust us ;-)

the problem you have is near with the USB to UART bridge and windows ;-)

classbproject, 2018-09-19:
@Harvie thanks! I know that you can change the baud in config.h but I remember reading somewhere that it shouldn’t be changed to anything less than 115200. I will look it up. 

@walterwismann thank you for all the testing. What hardware are you using? I will try changing the baud on my setup and report back. However, changing the comms setup for specific software seems inconvenient especially when it works with other softwares just fine.

walterwissmann, 2018-09-20:
No Problem ;-)

Arduino UNO at 115200 Baud:
[image]

The latest Version of Harvie's bCNC on RasPi 3+B.

In the very first time i have had also problems with an USB cable.
I take a other one, round 1,2m long, and it works. 

Walter

classbproject, 2018-09-20:
Ok let me try using another USB cable. The end stops on my hardware have an RC filter followed by a Schmitt trigger buffer. I know that the end stops are not causing this. If that were the case then we wouldn’t have seen a Serial error.

Harvie, 2018-09-20:
Are you sure? AFAIK in arduino there's no +12/-12V as it's TTL or 5V logic. I only had troubles with interferrence when i used unshielded spindle cable. need to say that i don't have limit switches installed as i found no need to use them. @classbproject have you tried disabling the limitswitches in grbl to make sure that they are not accidentaly triggered by EM interference?

walterwissmann, 2018-09-20:
Yesssss, i found it by my self and delete it ;-)

classbproject, 2018-09-20:
@Harvie No, I don't want to disable the limit switches. The machine is big and heavy and I don't want anything crashing when I make silly mistakes. I know that they are not being triggered. In any case, if they did trigger my controller has a visual indication of it being triggered and I see none. 

A quick update: I changed the baud to 19200. This now sets the baud clock divider of the CP2104 to a whole number. The only error now is that of the internal oscillator shift due to temperature etc. - factors that are anyway beyond user control. I've only tried running one instance of the program and it seems to run fine for now. This is with your version of bCNC.

Harvie, 2018-09-20:
I meant to disable limit switches temporarily just to check. but if you have visual indications, that's probably enough to be sure. But note that EMI would trigger them only for extremely short while, which will probably be unable to catch by eye as blink of the LED. unless you have some latching mechanism.

Is it possible for you to replace USB bridge for something else? Eg. FTDI ? Maybe cut traces and piggyback some FTDI module there.

Anyway this is certainly not bCNC problem. There might be some timing differences between various g-code senders, which cause CP2104 bug to not apply. But i think right solution would be to use chip which is not buggy.

classbproject, 2018-09-20:
The end stops on mine have a low-pass RC filter followed by a Schmitt trigger. This makes it extremely unlikely that they'd trigger even because of EMI. 

The RX and TX pins are broken out so I could try using a FTDI. The CP2104 is actually a great chip and I've been using it on other projects too without any problems. Adafruit, in fact, replaced their FT232's with this one on their Huzzah boards. I don't think the chip is buggy.

Harvie, 2018-09-20:
You can connect second USB serial bridge to TX pin of CP2104 to check it's output while sending to arduino.

Harvie, 2018-10-30:
Have you solved this?

classbproject, 2018-10-30:
@Harvie No. I didn't get the time to test your suggestion to see the CP2104 output. However, I did use UGS to run the same part and it ran flawlessly. I finally abandoned bCNC and finished all my parts on OpenCNCPilot and UGS. The fact that it ran on these softwares very well, indicates that there is no fault with the CP2104.

Harvie, 2018-10-30:
Can't reproduce.

closed 2018-10-30 by Harvie

christiankno, 2022-09-10:
I've been having the same issue. I have a file with around 200 000 lines of gcode. bCNC stopped after doing 6% of the job. 
The status line at the bottom said completed (6%) and the laser head stops moving but it leaves the laser on. 

It has happened with other files but since they have been smaller, i've been able to manually comment out the lines that were run and start it again so it continues with the rest. 

This time with so many lines of gcode, its not a viable solution.

Using LaserWeb to send the gcode with no problem shows the controller i
s able to handle this. But bCNC being so feature rich in comparison, and Ugs not even running in my Laptop, I would love to see a solution for this issue

Im running bCNC on Ubuntu 20.04

[TestCncWidthOverlap_Power75.gcode.txt](https://github.com/vlachoudis/bCNC/files/9539261/TestCncWidthOverlap_Power75.gcode.txt)
s able to handle this. But bCNC being so feature rich in comparison, and Ugs not even running in my Laptop, I would love to see a solution for this issue

Im running bCNC on Ubuntu 20.04

rschell, 2022-09-10:
Is the issue similar to #1726?  try adjusting the Timeout to a large number 120+ seconds and wait for the file load to complete.  If that works you will be the second one to be bitten by that "Feature".  Took my pi4+ 25 seconds to process your test file.

Harvie, 2022-09-10:
1.) Timeout only affects display on screen. G-code is always sent completely to CNC machine regardless of timeout.

2.) 99.9% of time the sudden problems during streaming were caused by EMI problems... (eg. stepper motors affecting USB cabling)

christiankno, 2022-09-11:
I didn't know that was the timeout for. glad to know this, since this file didnt even render completely which was frustrating me. So even if it doesnt solve this problem, solves another one i had. Thanks ill try it out later

If it was an EMi problem, wouldnt it be present regardless of the gcode sender im using?
 i feel like i could be missing something

Harvie, 2022-09-11:
> If it was an EMi problem, wouldnt it be present regardless of the gcode sender im using?

EMI is black magic. Each sender sends lines with different intervals and in different groups of multiple lines at once.
When i had these issues it even happened to me that only specific g-code sequences (moves) were triggering the EMI errors. Which makes sense, because g-code directly affects the frequency of power fed to the steppers.

But hey. We've also found software issues with streaming in the past. But first please make sure you don't have any EMI issues. Eg. try to stream the code to arduino without anything connected to it (eg. no stepper drivers, no spindle, no motor power applied.)
You can basicaly have arduino by itself with GRBL on your table without any additional devices connected to it and bCNC should be able to stream...