vlachoudis/bCNC #973: Stops sending

ErikLambrechts, 2018-09-21:
get this error 
```
Exception happened during processing of request from ('127.0.0.1', 54020)
Traceback (most recent call last):
  File "/usr/lib/python2.7/SocketServer.py", line 290, in _handle_request_noblock
    self.process_request(request, client_address)
  File "/usr/lib/python2.7/SocketServer.py", line 318, in process_request
    self.finish_request(request, client_address)
  File "/usr/lib/python2.7/SocketServer.py", line 331, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/lib/python2.7/SocketServer.py", line 654, in __init__
    self.finish()
  File "/usr/lib/python2.7/SocketServer.py", line 713, in finish
    self.wfile.close()
  File "/usr/lib/python2.7/socket.py", line 283, in close
    self.flush()
  File "/usr/lib/python2.7/socket.py", line 307, in flush
    self._sock.sendall(view[write_offset:write_offset+buffer_size])
error: [Errno 32] Broken pipe
```

walterwissmann, 2018-09-21:
Hi Erik,
can you send a little bit more from your problem?
Screenshot and or Files, please?

Sincerely Walter

ErikLambrechts, 2018-09-22:
Hey Walter, 

It seem to be a problem with the specific Arduino I am using. 
Thank you for the quick response. 

Erik

closed 2018-09-22 by ErikLambrechts