vlachoudis/bCNC #975: Drag Knife postprocessor plugin!

Harvie, 2018-09-23:
This does not seem very hard to do:

[image]

Harvie, 2018-09-24:
I've written first experimental dragknife plugin. Currently only works with straight lines:

[image]

sfinexer, 2018-09-25:
@Harvie 
On the first point there isn't enough radius how there will be the turn is made?

Harvie, 2018-09-25:
But you don't know initial position of the knife, so it's hard to manipulate it. You can add leading cut manualy. right now i don't know how to do it so that it does not accidentaly cut into piece. But i am going to rewrite this completely to support arcs. As i said, this is experimental plugin.

MARIOBASZ, 2018-09-25:
a possibility in which the first point is sea at a certain distance and in the opposite direction to the second point: if the first line of the figure goes from (0,0) to (10,10) it could be the new sea figure
(-2, -2)
(0,0)
(10,10)

Harvie, 2018-09-25:
I've done first test of the algorithm :-)
https://www.youtube.com/watch?v=egf1R4cKVlM

sfinexer, 2018-09-25:
@Harvie 
Incorrectly.
When the arch drawing not true is drawn.
Check for circles, you are convinced of it. The knife is late, it will turn out that an arch (and a circle) will be cut out by more smaller radius.

MARIOBASZ, 2018-09-25:
... give the initial point of sacrifice option outside / inside?

MARIOBASZ, 2018-09-25:
An idea:
1- locate coordinate and sacrifice surface
2- make the tool make a sacrificial cut parallel to the first real movement

sfinexer, 2018-09-25:
@MARIOBASZ, On straight lines everything is ideal.
I suspect them and tests were made ;)
Think we will bypass all proprietary software of the program!!!

Harvie, 2018-09-25:
> When the arch drawing not true is drawn.

@sfinexer thanks for testing. please send images of the problem!

> ... give the initial point of sacrifice option outside / inside?

it can be done, but it's not easy to make such algorithm work for any shape (you can have other parts of shape near the beginning, which will intersect the sacrificial cut). I think we can do this, but first i want to get shape cut nicely. beginning and ends can be tuned after the basic functionality is done.

> 1- Choose the coordinates and the sacrificial surface

you can just add aditional line to the beginning of shape. you can do this in librecad for now.

> 2- make the tool make a sacrificial cut parallel to the first real movement in that coordinate

If i were able to cut parallel with the first real movement, i would as well be able to cut the first real movement correctly :)

Harvie, 2018-09-25:
It seems to me that it's not really easy to process arcs for drag knife cutting. Some resources:
https://discourse.mcneel.com/t/drag-knife-toolpath-generation/25779/6
http://math.ucsd.edu/~ebender/87/bicycle.pdf

MARIOBASZ, 2018-09-25:
straight lines.
The sacrifice section, (if it is a good idea) is secondary to after you write the code. The idea is, if the first cut is not quite correct, that it does so in a part of the material to be discarded. If the picking is parallel to the first cut, it will align the cutter correctly at the start of the cut (s). A check box would give freedom if manually align or material sacrifice area

Harvie, 2018-09-26:
> Incorrectly.
> When the arch drawing not true is drawn.
> Check for circles, you are convinced of it. The knife is late, it will turn out that an arch (and a circle) will be cut out by more smaller radius.

@sfinexer do you think this is better?

[image]

toktuff, 2018-09-27:
I was thinking a little about this algorithm. The latest algorithm looks more robust.

Here is a outline of an alternative algorithm:
1. Compute shortest distance from endpoint of "seg" to the line formed by (pprev, pnext)
2. If this distance > dragoff - tolerance, subdivide pprev,pnext or replace with an arc.

This might be more robust then angle comparisons.

I hope this makes sense.

Harvie, 2018-09-27:
Does anyone here have machine with dragknife to test the toolpaths? I am in the process of making one, so i can't really test right now.

@toktuff Not sure what you mean. please send images :-) also check the algorithm i've pushed right now. it's much cleaner code!

toktuff, 2018-09-27:
I needed to rethink my idea a bit. The code does indeed look clean. It's the kind of thing that could have unexpected corner cases but none come to mind.

My only observation is the angle check 

`angle = degrees(abs( seg.tangentEnd().phi() - next.tangentStart().phi() ))`

This won't work if you end up at +-pi. A cleaner way to do it is to use the dot product of the two vectors and check if  dot(tangentEnd, tangentStart) < cos(thr) (if that makes sense)

Harvie, 2018-09-27:
@toktuff what do you mean by +- pi? can you please elaborate or send example of shape which will fail? i think you may be right.

Harvie, 2018-09-27:
@toktuff i've never used dot product, have to read about it. but i think it might be usefull to decide if i should swivel CW or CCW. Currently it's a hacky solution:

```
                                        arca = Segment(Segment.CW, opath[i-1].tangentialOffset(dragoff).B, seg.tangentialOffset(dragoff).A,$
                                        arcb = Segment(Segment.CCW, opath[i-1].tangentialOffset(dragoff).B, seg.tangentialOffset(dragoff).A$

                                        if arcb.length() < arca.length():
                                                arca = arcb
```

Harvie, 2018-09-27:
@toktuff now i calculate angle difference like this:
`degrees(acos(seg.tangentStart().dot(opath[i-1].tangentEnd())))`
thanks for idea!

But the angle is absolute shortest angle without sign, so i can't do this:

```
if angle >= 0:
       arcdir = Segment.CW
else:
       arcdir = Segment.CCW
```

Any ideas how can i easily calculate direction of rotation?

toktuff, 2018-09-27:
You probably figured this out by now but the problem is if since angles are periodic they have to wrap around some time. (Usually functions acos/atan etc return values in the range -pi..pi). So if you have a vector with an angle close to pi and another vector that is rotated a few degrees this could cause to wrap around the range.

Regarding the direction you could use a cross product (cross product is defined in 3D but just plug it into the formulas and let z be zero).

Harvie, 2018-09-27:
@toktuff i've been already doing some tests with cross product, but didn't came up with anything usefull.

toktuff, 2018-09-27:
Aah, sorry, for being brief. 

If you compute this value:
`z = a.x*b.y - a.y*b.x`
You can check the sign to determine if a is CW or CCW relative b.

This is also equal to:
`z = length(a)*length(b)*sin(angle)`
Where angle is the angle between a and b.

This is related to the cross-product. 
```
|ax|   |bx|   |      0      |
|ay| x |by| = |      0      |
| 0|   | 0|   |ax*by - ay*bx|
```

I hope this makes sense.

Harvie, 2018-09-27:
> `z = a.x*b.y - a.y*b.x`
> You can check the sign to determine if a is CW or CCW relative b.

i've been doing this exact thing and it failed in some cases and selected to longer arc instead!

toktuff, 2018-09-27:
That sounds strange. Perhaps there is something I'm not considering.

Harvie, 2018-09-27:
Sorry, i've tested it again and now it works! Probably had problem somewhere else.

Harvie, 2018-09-27:
Problem was that i was using  `if angle > angleth:` which was not triggered for negative angles lol.
Now i don't use angle sign, but i've added `abs(angle)` to prevent this in future :-)

Harvie, 2018-09-28:
I also added funny feature to evaluate the results. Dragknife simulator. You use it on dragknife toolpath and get back toolpath of the knife tip:

[image]

sfinexer, 2018-09-28:
He closes a trajectory.
The option is necessary to do it or not.
[image]

Harvie, 2018-09-29:
> On the first point there isn't enough radius how there will be the turn is made?

@sfinexer i've fixed this! now there's consistent entry/exit direction for all cuts

Harvie, 2018-10-01:
First real cut:
https://www.youtube.com/watch?v=fxP6KcZvIWc

sfinexer, 2018-10-01:
@Harvie How does the knife turn?

Harvie, 2018-10-29:
Merged into upstream

closed 2018-10-29 by Harvie

CCERocks, 2019-10-03:
Hi Harvie, great jobs with drag knife add on!... 

i wonder is it possible to make a drag rotary knife add on too ? :)

what i see it.. almost the same algorithm... or maybe someone test this drag knife with rotary knife. 

thanks.

Harvie, 2019-12-01:
This might be easy to add. Which axis is used to rotate the knife? Never used such device myself...

otoyasan, 2020-08-19:
こんにちは。
ドラッグナイフ、興味があります。が、私の環境でシミュレーションしようとしましたがGcodeに変化が起きてるように見えません。この機能についてどこかに詳しい説明などありますか？wikiでも見つけられず困っています。
私の環境は
win10home 64bit
cpu corei5 6200U 2.3Ghz
gpu cpu内蔵

bCNC0.9.14-dev(exe ver. もあります)

どなたかお知恵をお貸しください。
[image]

Harvie, 2020-08-19:
@otoyasan 
1. Uncheck the "simulate" checkbox. That causes the process to run backwards. Not good for cuting.
2. Enter your dragknife offset in mm (distance between knife tip and rotation axis)
3. Select the g-code using mouse (make it blue)
4. Click dragknife button at the side panel

otoyasan, 2020-08-19:
＠ハービー様

ありがとうございます！
こちらはもう深夜で私もアルコール摂取で手元が不安なので、明日起きてから早速実践させて頂きます！！

otoyasan, 2020-08-19:
[image]
酔ってても出来ました！！
ハービー様の的確なアドバイスに感謝します。

KjellVerb, 2022-03-08:
Hi @Harvie , thanks for the great plugin. One thing I'm wondering about: is it possible to have multiple toolpaths separated by lift moves? I'd like to create vinyl lettering so each letter is a separate toolpath and they should be connected into a single gcode by moves with up/down motion in between.

Harvie, 2022-03-08:
> separate toolpath and they should be connected into a single gcode by moves with up/down motion in between.

Isn't that possible right now? I beleive there was a way, perhaps split the input paths to separate blocks before applying plugin?

KjellVerb, 2022-03-08:
Ah got it, I originally applied the plugin to an imported gcode but when I import the original svg instead, it does what I expected. Only disadvantage is that it now uses many small straight G1 moves for rounded letters instead of G2/G3 arcs. Maybe that's due to my bCNC settings and not due to the plugin, I'm still learning. At least I get usable output now, thanks!

KjellVerb, 2022-03-08:
Hmm I think I've found two new issues but I hope I'm wrong :-)
I see a workaround for the first one but for the second I think I'd need help to solve it. 

1) Internal geometry (like the hole inside an "O") is ignored by the script if it's in one block together with the enclosing geometry. Workaround is to select the enclosed geometry in the "Editor" tab and cut/paste it to a separate block.

2) For multiple geometries in a single workspace, the plugin always starts with the assumption that the knife is in the initial direction that was entered as a parameter. This is usually not the case in the real world, instead the script should somehow remember the tangent direction from before the previous lift move and apply that as the initial direction for the start of the next toolpath. The only workaround is to add "pause" gcode and manually change the knife direction every time but that's not really desirable.