vlachoudis/bCNC #978: Scaling!

sfinexer, 2018-09-25:
I suggest to add a scaling option. That it was possible to change in proportion the object size (X-Y) or (X) or (Y) or (Z) or (X-Y-Z)

It is very useful when there is a certain initial object!!! and it is necessary to make similar but other size!

MARIOBASZ, 2018-09-26:
First images of the scalar plugin
Original circle:
[image]
Scaling 2:1 in xy plane
[image]

2: 1 scale in X direction; scale 1.5: 1 in Y direction
[image]

Notes:
1-They are all straight sections, there are no curved lines.
2-The idea was to make a plugin for backlash problems, and, using the same concept, modify it to make a plugin to scale. I have reversed the steps, with a small modification I will upload a plugin for backlash (maybe I will also add it to correct orthogonality problems)

closed 2018-10-13 by sfinexer

jpagliaccio, 2019-12-04:
I love the scaling - it's very useful. In version 0.9.14-dev, the new scaled path is all G0 rapid. Am I doing something wrong? My scale settings are X Scale = 2, Y Scale = 2, Z Scale = 1.

MARIOBASZ, 2019-12-04:
I'm glad you find it useful!
I do not remember modifying something, I will review it.
Maybe the depth z is equal to or greater

[image]
 than zero? (I am very bad in memory, I do not remember well as I did)
In my machine I have made another pado, from a circle I came to this:

MARIOBASZ, 2019-12-05:
The problem is as follows: If the gcode to be scaled is below the surface it generates g1, but g0, the reason is that on the surface the displacement is with g0