vlachoudis/bCNC #983: Spiral cut

MARIOBASZ, 2018-09-28:
I have written some lines of code for a complement, which allows in an alternative the spiral cut. The last pass is circular:
[image]

What I write is called a splice generator, the variables to enter are:
center 1, center 2, radius 1, radius 2, angle 1, angle 2 (angle is separated by 2: phi and alpha 1 and 2 respectively, using the trochoidal concepts of @harvie plugin). It allows other alternatives that only Arquimedes spiral

sfinexer, 2018-09-28:
It is ingenious, it is so possible to make deepening a thin mill!

MARIOBASZ, 2018-09-28:
Use as soft splice of curves
[image]

curiosity or use as ... peacock tail ??

[image]

MARIOBASZ, 2018-09-28:
@sfinexer , Yes, I find it useful in my work (I'm going slow) version of trochoidal cut: I descend in helix and then enlarging the hole.
It could also be another alternative for pocket operation (especially at high speeds to smooth out jolts in passages from one diameter to the next as it may happen now) or a small contribution for adaptive cuts.