vlachoudis/bCNC #984: Question Intersection, union, problem difference. Possible exclusive union?

MARIOBASZ, 2018-09-28:
had not seen the intersection and difference utilities @harvie.
I have drawn two squares.

[image]

1- If I choose to make a difference between both squares, two figures appear identical to the selected squares, I do something wrong or it is not working difference.
2- Intersection repeats the common routes. Minor detail, but maybe you can make the route a single one instead of a duplicate one.

[image]

3-Possibility of implementing exclusive union?(I do not know if I had already asked him sometime) (intersection appears with dots)

[image]

MARIOBASZ, 2018-09-28:
I began to read the code, I lack basic knowledge of some things (can ask things things if I lack to be able to understand?). The difference works correctly to subtract island

MARIOBASZ, 2018-09-28:
rying to adapt for external union or exclusive union (that from point A of intersection to point B of intersection the road is by the outside), I discovered that if one of the figures starts inside the other, intersection has a result other than intersection:
[image]
It would not be a problem for external union

Harvie, 2018-09-29:
These plugins are in development. They are only building blocks for bigger plugin for pocketing. They are not meant to be used directly for now!

MARIOBASZ, 2018-09-29:
Sorry, I had not noticed.
Woww !!! You are a few steps to use on the PCB islands!

[image]

It would be an excellent strategy to be able to use a combination of two or more endmil for the elimination of copper

Harvie, 2018-09-29:
I don't think it makes sense to use bCNC as PCB CAM right now. There is very nice free opensource program with better fit for this task:
http://flatcam.org/manual/procedures.html#copper-area-clear

MARIOBASZ, 2018-09-30:
I have used Flatcam before, I prefer pcb-gcode. The good thing for me would be that from pcb-gcode you could use more than one endmill for the widening of copper-free zones, instead of successive passes with an endmill of 0.10, or from here, which I still believe can be a simple option, quick to use and very adaptable by the intervention of different opinions. After all, you have been criticized for trying an adaptive milling existing 360, and at least we both went to trochoidal milling (by the way, you still have at your disposal the diameters for adaptive trocoid in the corners), upload the spiral machining to enlarge the holes, there is the descent in ramp, helicoid ...
And with helical drilling, adapting it, (an idea pending to implement) with the same tool to perform the different perforation diameters, it is possible to optimize the PCB operations with the least change of tools.
I do not know how to do it, for you maybe it's very simple: Having the points of intersection A and B, can you generate the route so that the result is from the outside, instead of from the inside? (the opposite way to the intersection) There would be a "union for the outside" or exclusive union.

Harvie, 2018-09-30:
@MARIOBASZ i know. but there's much more to milling PCBs. it's whole rabbit hol. starting with parsing gerbers. I think this is not gonna happen soon.

MARIOBASZ, 2018-09-30:
it is not necessary to understand Gerber, only to handle figures, and the restriction to enlarge is like the posters "Prohibido Ingresar. Private property. The sentry will open fire "
or "Beware of the dog. You can piss her shoes "

closed 2018-10-04 by MARIOBASZ