vlachoudis/bCNC #988: Plugin for PCB from Eagle?

MARIOBASZ, 2018-09-30:
Chilipeppr has a widget to do the work on pcb reading directly from the Eagle .brd Is it possible to implement something like this for BCNC?

Harvie, 2018-09-30:
Aren't the gerbers standart format to pass PCBs to production rather than eagle brd?

MARIOBASZ, 2018-09-30:
In Chilipeppr you save the intermediate step by working from the .brd file that it generates automatically in Eagle. With the pcb-gcode plugin inside eagle, it is also very comfortable, since the parameters you want are set (speeds, tip diameter, position for tool change, step over and number of passes) in a very automatic way (so simple as just pressing a button). To my liking, the improvement would be: intelligent passes according to diameters of end thousand in the elimination of copper: I populate the following, suppose 0.8 as the maximum cleaning chosen, and endmils to use bit 1 = 0.10 and bit 2 = 0.7, forcing a pass mandatory with bit1, in many areas with a single pass with bit2 would be ok, cleaning is left in specific sectors with bit1 .
Another option is that the first pass is as it comes according to the chosen program Flatcam, pcb-gcode, or do not know what to do other than eagle. Being able to use an overcut strategy for different end mils is what would be a big step

MARIOBASZ, 2018-09-30:
https://github.com/chilipeppr/widget-eagle

MARIOBASZ, 2018-11-16:
Advances. Pending to automate actions.
Example: in an isolate with bit 0.1 (origin: pcb gcode for example)

[image]

Isolated extra with bit 0.7.
Where this bit does not arrive, pocket operation with bit 0.1
(in two passes, +0.35 is isolated, more passes in pocket sectors, instead of 7 passes)
[image]

MARIOBASZ, 2018-12-03:
https://github.com/JohnAtl/pcbgcode/issues/2#issue-386977226