vlachoudis/bCNC #989: Add new parameters?

sfinexer, 2018-09-30:
I remade GRBL as me in b-CNC to add new parameters $140 $141 $142?

sfinexer, 2018-09-30:
GRBL has a shortcoming associated with homing_pulloff.

And if, on one of the axes, the maximum displacement is less than the necessary movement to disconnect the end sensor of the other axis.
Then I can not use the homing cycle.

In other words, homing_pulloff or not enough to open the end sensor for one axis, or more than moving on the other axis.
This is possible in the case of a long one axis, but too short the other axis of the machine.

MARIOBASZ, 2018-10-05:
Sorry, I did not read you. I also grieve when I make a query and I do not receive help, (sometimes, sometimes yes and thank you) even more when my intention in general is to upload some new utility. By virtue until now I have been able to continue advancing using contributions made by others. $ 140 to $ 142 did you upload it in grbl? If so, we must upload it to BCNC.
I notice then how and where I have uploaded it for a development that I am doing (an automatic change of tools of reduced cost), if it is not provided by someone before the solution. By the way, two things: I saw that you had a problem when in a code the distances "i" and "j" were wrong for a C2; I have not put with my CNC, the problem was that it did not leave the notice that the error is due to bad values ​​in C2 or C3? If so, you would have to ask Harvie if you can restore the warning notices that by mistake have been disabled (I referred to this in another issue). In a previous copy, these notifications work correctly.
The second is that I read to you about what you wrote before. Maybe I need help to complete an optimization of the milling process, using, (at least for now) two dustintos diameters for the emptying of copper. Do you have knowledge and may interest you to help me? I extend the restlessness to the rest of the community.

sfinexer, 2018-10-05:
@MARIOBASZ 
The bottom line is, $ 10 = homing_pulloff
And it is the same for all axes.
I am for OUR GOALS (I know the size of the part) so that when searching for the starting point, the machine automatically positioned at the beginning of the workpiece.
In other words, I have introduced three new parameters in GRBL for my needs.
settings.c line 60+
```
.homing_pulloff_new [X_AXIS] = DEFAULT_HOMING_PULLOFF,
.homing_pulloff_new [Y_AXIS] = DEFAULT_HOMING_PULLOFF,
.homing_pulloff_new [Z_AXIS] = DEFAULT_HOMING_PULLOFF};
```
I need this in order to immediately move the machine to the zero coordinate of the header when searching for the starting point.

I will fix your scaling.

MARIOBASZ, 2018-10-05:
I have corrected scale, you can choose several blocks. Upload the modifications as before (unless you tell me otherwise for possible conflicts in Guthub, I have asked this question about which path to follow)

sfinexer, 2018-10-06:
In the video:
http://cnc.tiven.ru/ssr-2018-10-06_08.51.44.mp4

sfinexer, 2018-10-06:
Where in b-CNC code is responsible for reading and writing the configuration of the controller?

coilheaddesigns, 2020-02-07:
where can i find this grbl version, would like to have independent pull-off per axis PLEASE