vlachoudis/bCNC #992: Job does not display

micksulley, 2018-10-02:
I have bCNC installed on a Raspberry Pi connected to my router and it works fine.
I have also installed bCNC v 0.9.14 on my desktop (Mint19) and it runs but does not display the job, just the co-ordinate arrows.  I have zoomed in and out, I have tried X-Y, X-Z, ISO1, etc, the co-ordinate arrows display as expected but no job. If I use the editor tab I can see the code, so the file is loaded, but doesn't display.  
I have run it from a terminal and there are no errors displayed.

Any ideas?

Harvie, 2018-10-03:
Have you tried latest git version?

sfinexer, 2018-10-03:
@micksulley controller $10=3 !!!!

sfinexer, 2018-10-03:
video
http://cnc.tiven.ru/ssr-2018-10-03_08.05.50.mp4

micksulley, 2018-10-03:
Just downloaded from GitHub and it is the same version  that I am running.
I am not actually connected to anything on my desktop, but want to use it for editing g-code, so it would be really useful to see what I am doing :)
On the Pi when I open a file it displays the job outlines and paths even if it is not connected, but on the desktop it doesn't.
The $10 parameter as I understand it controls view of current position of the cutter but doesn't affect display of the job, is that correct?

Harvie, 2018-10-03:
[image]
Do you have these two buttons enabled on top of the display?

Harvie, 2018-10-03:
Can you send us the g-code file?

MARIOBASZ, 2018-10-03:
1-Maybe the canvas is showing a position away from the work (fix by putting the mouse over to see if you're in the right place)
2- as you were told, press the buttons to visualize movements G0 and G1

Harvie, 2018-10-03:
1.) Fit to screen button:
[image]

sfinexer, 2018-10-03:
@micksulley
Parameter 10 transfers the current machine position (coordinates) to b-CNC.
If it is 0, it will be as I have on video.

micksulley, 2018-10-03:
Fixed!  It was the 'Toggle Display of Paths' button.  
Many thanks to all for your help.
Mick

closed 2018-10-03 by micksulley

Harvie, 2018-10-03:
@micksulley for you it was disabled by default after install? or you accidentaly disabled it?

micksulley, 2018-10-03:
I can't be sure.  I certainly didn't disable it on purpose, can't see why you would want to, so I assume it was disabled by default.
I downloaded and installed a new copy from GitHub this morning to be sure I was on the latest and it was off then.  Would that set it to default or is there a config file which would remain from last time?

MARIOBASZ, 2018-10-04:
... but if preloaded $ 10 = 0, the problem comes from the default setting of grbl, in the new version then? It would be good to mention it there: I would have gone crazy thinking what I did wrong or what is wrong if it were the first time I downloaded the program

micksulley, 2018-10-04:
OK to sort this, I did not have bCNC on my laptop, just downloaded and installed it, opened it up and opened a g-code file and it displays the job.  So it does seem that I somehow turned off 'Toggle display of rapid motion' by accident.
Sorry!