vlachoudis/bCNC #994: bCNC uses cnc 100% cpu and never finish pocket operation when gap and endbit is exact same size

DaveDavenport, 2018-10-04:
The attached model 
[test.zip](https://github.com/vlachoudis/bCNC/files/2446755/test.zip) if you set endmill to 2 mm it never finishes the pocket operation.
It uses 100% cpu and you need to kill bCNC. 

If you set it to 1.999mm it works fine.

(tried this on latest git.)

walterwissmann, 2018-10-05:
Hello Dave,
[image]

The min space between the tracks are 2mm ...
it can not work with 2mm endmill ;-)

Harvie, 2018-10-06:
I will completely rewrite the profile/pocket code soon, so this problem will most likely not apply to the new code.

DaveDavenport, 2018-10-07:
I realize it won't work, but it completely hanging the gui is not good.

MARIOBASZ, 2018-10-15:
It is a bad decision to correct, to hang the program. Could you make it alert with alarm and not allow the plugin to walk if the space is insufficient?