vlachoudis/bCNC #998: ArcFit does not work

sfinexer, 2018-10-09:
I bring g-code and video.
g-code:
[test.nc.zip](https://github.com/vlachoudis/bCNC/files/2462017/test.nc.zip)
video:
http://cnc.tiven.ru/ssr-2018-10-09_23.13.37.mp4

Harvie, 2018-10-11:
[image]

This is too mesy shape and arc fiting will probably never work for this. Also 10mm precision is too 
much. You should stick with something like 0.5mm.

But you are true!

I've just found some bug in arc fitting algorithm, but i cannot figure out how to fix it. If you wish to look at it in lib/bpath.py. It's the code which checks whether two subsequent segments steer left or right (because in arc all segments should steer in one direction)

```python
        def arcFit(self, prec=0.5, numseg=10):
                def arcdir(A,B):
                        #FIXME: here is some bug. A and B is not used at all, but it works better this way. Have to figure out what's going$
                        TA = tmpath[0].tangentEnd()
                        TB = tmpath[1].tangentStart()
                        if (( TA[0] * TB[1] ) - ( TA[1] * TB[0] )) < 0:
                                return 1
                        return 0
```

The two buggy lines should look like this instead:

```python
                        TA = A.tangentEnd()
                        TB = B.tangentStart()
```

However it stops working completely when i fix it! That means that there is some another problem!

sfinexer, 2018-10-11:
I checked the work of the new function, and I saw that there was a mistake somewhere.
I am programming on python for the first time, I am used to C/C++.

sfinexer, 2018-10-11:
Harvie, the more important issue I indicated:
https://github.com/vlachoudis/bCNC/issues/997

There is clearly somewhere a mistake in the program, which does not allow to work with negative coordinates.

Harvie, 2018-10-13:
I've fixed the problem. Now i sucessfuly fitted arcs onto your g-code. But as i said... because your path is so messy, i can only arcfit 1-5% of it. Here is example of such arc:

[image]

Harvie, 2018-10-14:
> I think that if the path is bad, then we must ignore it, rather than build arcs.

But that's exactly what happens. That's why only 3% of your path gets arcfited. Because there's no fiting arc for the rest. That means that 97% of your path is ignored. So i am not sure what you mean. Today i've added help to make purpose of this plugin clear.

There was bug that produced that funny circles all over the place (like in your video). But that's no longer the case i've fixed that yesterday.

Harvie, 2018-10-14:
I made some improvements and i've managed to reduce your sketch almost to half:
[test-mrg.nc.zip](https://github.com/vlachoudis/bCNC/files/2476920/test-mrg.nc.zip)

While preserving looks:
[image]

Harvie, 2018-10-14:
Now i've reduced to third size with the same settings and better algorithm:

[test-mrg2.nc.zip](https://github.com/vlachoudis/bCNC/files/2476940/test-mrg2.nc.zip)

[image]