vlachoudis/bCNC #999: GRBL Tool Change/Calibrate Probe Fail

sarf2k4, 2018-10-12:
Hi,

I am new with bcnc and using cnc machine itself as well. I am currently using latest bcnc v0.9.11 and just finished setting up my z probe for easier TLO calibration later. No matter what option I put on the Probe>Tool either G38.2 or G38.4, I always get this message "ALARM: Probe fail" and the latest gcode that triggered the failure is `g91G38.5f100.000000z5.0`.

I personally picked `g38.3` option for this tool calibrate/tool change and from the terminal, below is the important gcode command regarding the probing
```
g91G38.3f100.000000z-5.0
[PRB:0.000,0.000,0.088:1]
ok
g91G38.5f100.000000z5.0
ALARM: Probe fail
```

If I am correct, I don't really need that g38.5 which is stop on loss contact, because it is the one giving me this error. Even through terminal I copy pasted the g38.5 line I still encounter the error. Is there a way to have bcnc not send G38.5, but increase the z height instead?

P.S
Even tried controller grbl-v0 and grbl, both are the same

toktuff, 2018-10-12:
I would expect it to work. Perhaps there is something subtle here that I'm
not thinking of.

Anyhow, you should be able to use a simple g0 command to back of the proble
plate. I.e. replace G38.5 with G0 (and remove the f100 since g0 doesn't
need a feedrate).

I hope this helps and that I understood the problem correctly.

sarf2k4, 2018-10-13:
Hi, 

Actually, G38.5 will always be injected right after G38.2 or G38.3 automatically for tool change in stock bcnc interface

walterwissmann, 2018-10-13:
I think that is a problem with the Probe-Pin:
https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.9#6----probe-pin-invert-bool
> $6 - Probe pin invert, bool
By default, the probe pin is held normally-high with the Arduino's internal pull-up resistor. When the probe pin is low, Grbl interprets this as triggered. For the opposite behavior, just invert the probe pin by typing $6=1. Disable with $6=0. You may need a power cycle to load the change.

sarf2k4, 2018-10-13:
I checked the probe pin invert through bcnc, it is unticked. If I understand correctly about this one, if ticked, it will stay high, which will render g38.2 g38.3 useless which is required during the toolchange and probe. It may work with g38.4 and g38.5 but I think, we seldom use g38.4 and g38.5.

Also at tool change calibrate, g38.5 will be inserted automatically as what I stated in my previous post. I just wanted to prevent bcnc sending g38.5 with the probe fail result

Harvie, 2018-11-01:
So everything is working for you now?

sarf2k4, 2018-11-01:
No, what I did is to divide the nc files for each tool

Here is the screenshot of my bcnc alarm: probe fail
The picture below is that I pressed the "change" button on the toolbar
[image]

While the following is the terminal output from bcnc to the grbl firmware
[image]

Harvie, 2018-11-01:
Please install latest git version of bCNC. It has "[P]" letter indicating if probe is active or not. That way you can easily check if everything is working as expected. Note that probe has to be inactive before you start probing.

sarf2k4, 2018-11-01:
downloaded the latest bcnc and I can't launch it with "bcnc.py", error with numpy

Harvie, 2018-11-01:
new version of bCNC needs numpy module installed.

sarf2k4, 2018-11-01:
i have numpy installed, 
[image]

Harvie, 2018-11-01:
I don't know much about windows, but this seems to me that numpy is not installed correctly. Maybe you have multiple python versions installed and numpy is not installed under right version of python?

sarf2k4, 2018-11-01:
ok... I got the latest bcnc running, apparently I had to navigate to `python27\scripts` and execute this command `pip2 install numpy`

sarf2k4, 2018-11-01:
I tried with the latest bcnc update, still having the same problem where G38.5 is injected

The following 2 pictures is where I pressed the "change" button from toolbar
[image]
[image]

While these 2, I pressed the "calibrate" button
[image]
[image]

I tried several times to search for the injected g38.5 for these operation from the python script, but I don't think I'm able to read the script

Harvie, 2018-11-01:
> Where can I find this [P] thingy?

[image]

sarf2k4, 2018-11-02:
I didn't notice the "[P]", does that mean it will show if the probe is active high/triggered? I may have overlook at it

Furthermore, I did the wiring ok, I soldered the probe pin and connected to a wire>aligator clip>metal plate for probing purpose. 

I eventually tried yesterday inverting the probe pin `$6=1` but results is just like I predicted rendering G38.2 and G28.3 had errors.

Though I am still new to cnc machine software, l tried Universal G-code Sender, grbl controller and this bcnc. I personally like bcnc because it is intuitive and its flexibility. 

May I request that you try out the toolchange module and see what really is going on here. As far as I know, there are no other free and open source cnc machine software that can filter out M6 command

Harvie, 2018-11-02:
> I didn't notice the "[P]", does that mean it will show if the probe is active high/triggered?

yes. [P] appears when probe is active

probe is active when pulled LOW
if you invert probe it is active when NOT pulled LOW
arduino have only PULLUP, so you always have to connect probe pin to GND in order to read probe state.

sarf2k4, 2018-11-02:
My cnc machine is a cheap chinese desktop cnc primarily used for pcb milling, the motor and all of the frame act as ground terminal, in my previous post, I did say I soldered a pin on the probe pin I think A5, also I checked the pin trace location before I soldered it last time. What I have to do is to connect the probe pin to a metal plate, let the motor's tip touch the probe plate to trigger the event.

But main issues right now is bcnc added G38.5 after G38.2 or G38.3.

einencool, 2018-11-05:
Since I updated bCNC so that I can start my Code by pressing the Hardware "start" Button, I have many problems with probing my milling cutter. It often gives me a alarm without a message and stays down on my probing pin in contact.

Normally my cutter drives down onto the pin, releases a few 0,xmm and then probes again with low speed. 
But now it only drives to contact and stops with an alarm.

This is my command for probing:
g91
g38.2 z-5 f50
g0 z0.15
g38.2 z-0.3 f1
g90
G10 P1 L20 Z29.158
g0 z30

It works for many month without any Problem, and with the new Version of bCNC there are many failures. In most cases it works on the second attempt.

Greets Chris