vlachoudis/bCNC #1000: Scaling Does not work

sfinexer, 2018-10-13:
If you select multiple blocks, the scaling does not work.
Only the first selected block is executed.

MARIOBASZ, 2018-10-13:
ou can see the new extraction request if it works correctly. Scales the selected blocks. The scaled blocks are located before the last original block (point to be corrected). I must add the option not to scale in positive Z

MildaGenius, 2018-11-16:
Iam actually looking for scaling possibility, but cant found it in bCNC editor.

Can you please point it me, where scaling should be?

Thanks

MARIOBASZ, 2018-11-16:
In the last version I have not found it.
I have conflict in my github branch branch, I hope if someone helps me fix it, I can upload the updated scale plugin to select different blocks. If you want to scale, I think I've shared the plugin file and the icon file to make it work, and upload them manually. Or I can share it again.

MildaGenius, 2018-11-16:
If you can share it, I will try to install it manually.

Thanks

MildaGenius, 2018-11-16:
I made scale test and it looks like plunge feed is not set to generated GCode. For plunge feed is used cutting feed. Is this bug or do I something wrong?

[image]

[image]

[Bell_390_100_3_downscaled.txt](https://github.com/vlachoudis/bCNC/files/2590732/Bell_390_100_3_downscaled.txt)

MARIOBASZ, 2018-11-16:
I will check it. Surely my mistake. I think I have not uploaded to my branch the latest version

MildaGenius, 2018-11-25:
Any news about upgraded scaling plugin?