vlachoudis/bCNC #1001: Curved portal.

sfinexer, 2018-10-14:
I think that it is necessary to enter the portal curvature function in the settings (or options) of b-CNC.
If the portal is not straight, but sags in the middle (under its own weight), adjustments must be made in the g-code.

MARIOBASZ, 2018-10-14:
Interesting.
I have not used Slicing 3d, does this solve it?
Imitate auto level or something like that, can be a way to go.
If it remains as a predetermined compensation, there will have to be a cancellation of its use with auto level, or to compensate the obtained data.

sfinexer, 2018-10-14:
I drew the possible variants of the portal in red.
It can be bent by its weight.
I think this should be entered into the settings of the machine.
http://www.cnc-club.com/forum/viewtopic.php?f=15&t=20385

http://www.cnc-club.com/forum/viewtopic.php?f=15&t=20385&sid=825bb7b9bf85c8591abdc28b90cdbfb0&start=80#p462253

MARIOBASZ, 2018-10-14:
If you make a flat cut in x compensates the height Z according to the position.
Then, for the self-leveling probe, you must compensate inversely the established values. Or give option "do not use the portal curve". Look at issue # 926, modifying the plugin in process, it's a similar idea

MARIOBASZ, 2018-10-14:
Are we talking about the same. must compensate the value of Z so that the result is flat according to the position on the X axis

MARIOBASZ, 2018-10-14:
The glass plates come completely flat or must a self-leveling also be done?

sfinexer, 2018-10-14:
@MARIOBASZ 
Well, yes, the glass is flat, it is necessary to compensate for the inaccuracy along the Z axis, depending on the position of X.

MARIOBASZ, 2018-10-14:
You're right. I was reasoning wrongly. It is a good idea to compensate Z distortion in different positions with firmula or point clouds according to comparison with perfectly flat reference surface

sfinexer, 2018-10-14:
@MARIOBASZ That's right, Exactly!!

Harvie, 2018-10-14:
In this video he uses bCNC to mill on very uneven surface and it still works perfectly:
https://www.youtube.com/watch?v=dgqRBbZXwIA

sfinexer, 2018-10-14:
Ingeniously!
I did not know.

MARIOBASZ, 2018-10-15:
If the guides are irregular or the portal is curved, there is currently no solution for this: you would have to make a first auto level in negative to correct the imperfections of the machine: Suppose the portal were concave, there should be a convex machine corrector for that when sending a flat milling code it does not copy the concave deformation to the material.

MARIOBASZ, 2018-10-15:
Correcting guide, quadrature and backlash imperfections (these two problems I think I can correct them), a very economical machine like my first cnc I can improve the benefits!

Harvie, 2018-10-15:
> For surface autolevel, the two compensations should coexist: The universal for the machine plus the individual for the material.

This does not make too much sense. If you autolevel the material, it will automaticaly account for machine imperfections, because you use the imperfect machine to probe the material. In such case there is no need to autolevel the machine separately.

If you need to autolevel only the machine and you don't mind that material itself is not level (most time it is not). You can probe against something flat like your wasteboard or glass. That way you will get autolevel data for you machine alone not accounting for the material.

You can save these two as separate files. One for machine+material and one for machine itself. and load the one you need.

If you need to use the probe data of material on another machine, you can subtract the machine autolevel data from the machine+material autolevel data. This can be done using bCNC.

> If you correct defects in non-rectified guides, curved portal, square imperfection and backlash

IMO backlash should be corrected in motion controller on step level not in bCNC.

MARIOBASZ, 2018-10-15:
How silly can I be sometimes! True, it makes no sense to make two compensations! If I do not reason wrong this time, to compensate for the bad heights in the machine, it would have to compensate by means of a negative of the values of autolevel when executing on a flat surface of reference, to be able to give flat termination compensating the hardware defects.
Backlash Theme: When I knew less than now, I did not get backlash response in grbl problems. Also, it is the right place to make the correction. As usual, there is more than one possible solution to a problem. Making a complement today is the easiest solution to do for me.

closed 2018-10-15 by sfinexer