vlachoudis/bCNC #1002: Dots pictures.

sfinexer, 2018-10-14:
When I make pictures with dots, I had a clear positioning. I made a transformation for the random arrangement of points in a given interval from the initial one.
So the picture looks alive.
https://github.com/Sfinexer/bCNC/tree/sfinexer-NEW
http://cnc.tiven.ru/ssr-2018-10-14_16.07.12.mp4

Harvie, 2018-10-14:
So what's the issue?

sfinexer, 2018-10-14:
@Harvie No problem, this is a new option.

Harvie, 2018-10-14:
So why you create new issue for something that is not issue for you? :-)

sfinexer, 2018-10-14:
@Harvie I want to know how relevant this opportunity is.

Harvie, 2018-10-14:
I am not sure how many people will use this. However there's no reason to not include it. Maybe put it in "artistic" cathegory, so it gets hidden deeper into the menu.

closed 2018-10-29 by Harvie