vlachoudis/bCNC #1004: Scheduled assignments.

sfinexer, 2018-10-14:
There is an idea to implement the implementation of a certain task on a schedule. For example, to make 500 blocks of code, and after that, automatically execute a certain "Use configuration button" command, or a similar one.

This option is required to clean the stinger after several cycles of operation.

MARIOBASZ, 2018-10-14:
Or count while z <= surface, or send some signal from bcnc

sfinexer, 2018-10-14:
@MARIOBASZ I have too little python programming experience.
I will do as much as possible.

MARIOBASZ, 2018-10-15:
I am not a programmer, until recently almost no knowledge of python and currently with a lack of knowledge, and I have been able to move forward. Luck.
Idea: Use distance traveled by the tool instead of working hours. I think this value is simpler to calculate. And if you use a matrix "distance, material" I think you can have a better estimation of wear of the edge of the tool.

sfinexer, 2018-10-15:
@MARIOBASZ I think you are right.

Harvie, 2018-10-30:
This sounds like perfect way to crash your machine, when you forget that you've enabled such feature.
I think you should add this to g-code. You can use % python evaluation to call some python procedures from g-code.

sfinexer, 2018-10-31:
@Harvie To add a G-code transformation option in the form of a plug-in.