vlachoudis/bCNC #1005: The source code is not clear.

sfinexer, 2018-10-15:
The b-CNC code files are too large, not to mention the fact that many files are not used, for example bCNC-master/lib/python_utils/__ about__.py

It is impossible to understand what part of the code is responsible for what!
I need to find the cycle sending commands for grbl where is it ????????

sfinexer, 2018-10-15:
It is necessary to clean the code.
Many files are unnecessary, and many (most) are too overloaded with code.

Harvie, 2018-10-17:
Don't worry about lib/python_utils/__ about__.py :-) it's not really part of bCNC, it's just 3rd party library that was included into bCNC. there's no point in cleaning it because it will eventualy get back when we update to latest version of that library.

> It is impossible to understand what part of the code is responsible for what!
> I need to find the cycle sending commands for grbl where is it ????????

Have you tried `Sender.py`, it's right in the project root. It seems like quite clear name to me.