vlachoudis/bCNC #1010: On Ubuntu 18.04 no icon in the dash

FabArd, 2018-10-19:
Hi,
I am on Uuntu 18.04 LTS up to date and the python version install is : Python 2.7.15rc1

When I execute bCNC no icon appear in the dash. The place where must be the bCNC icon is empty and when I put the mouse on it there is "Toplevel" instead of bCNC.

[image]

sfinexer, 2018-10-20:
I collected all the branches, try this:  https://github.com/sfinexer/bCNC

FabArd, 2018-10-20:
This is the solution : 
In bCNC.py, I made this changes : 
103 else:
104       #FabArd self.iconbitmap("@%s/bCNC.xbm"%(Utils.prgpath))
105       #FabArd 1 for bitmap icon
106       #FabArd self.iconbitmap("@bCNC.xbm")
107       #FabArd self.iconmask("@bCNC-mask.xbm")
108       #FabArd 2 for image icon
109	     icon=PhotoImage(file="bCNC.png")
110       self.tk.call('wm','iconphoto',self._w, icon)

If you choose bitmap icon, you need to export "bCNC.png" whith gimp whit the option "generate mask file". Whit the second solution you can use color icon.

[image]

FabArd, 2018-10-30:
This is the problem on Ubuntu 18.04 and 18.10 : When I launch bCNC, there is an empty square in the dock where the icon should take place. When I place the mouse on the icon, the balloon show "Toplevel" and not bCNC. On the apps bar on the top of the screen, there is also "Toplevel" instead of bCNC

[image]

I made a test on Lubuntu (a light version of Ubuntu) using LXDE for the GUI (instead of Gnome for Ubuntu) and there is no problem : the icon is shown correctly and the balloon when you put the mouse cursor on the icon is "bCNC".

Harvie, 2018-10-30:
Problem is that .png support in tkinter requires PIL, which is not always available on other platforms. That's we have to use GIFs. Does it also work with GIF ?

FabArd, 2018-11-01:
I installed the package python-pil and made a test with GIF icon : nothing is running. I execute bCNC and nothing appear on the screen.

Harvie, 2018-11-01:
Why you close this issue then? We can try to fix it somehow.
I did not suggested you to use PIL. I suggested that we have to find solution how to get this working with .GIF icon.

FabArd, 2018-11-03:
Sorry, I did not realise that I closed this issue. I made a mistake by clicking on the wrong button : sorry !!!