vlachoudis/bCNC #1013: ERROR!!!!!!!!!!!!!

sfinexer, 2018-10-20:
video
http://cnc.tiven.ru/ssr-2018-10-20_11.34.13.mp4

MARIOBASZ, 2018-10-20:
Accidentally boosted warnings with indications or explanations of type of error occurred in the version of @harvie. It is desirable to correct this. Surely the error occurred in the wrong parameters for G2 / G3 (ij), probably due to rounding (I have an older version of BCNC in my PC, using it has helped me find the errors) If you look at the terminal, you should see the error number (example error 22)

sfinexer, 2018-10-22:
@Harvie Everything is very clear there. And then what is the mistake ???
http://cnc.tiven.ru/ssr-2018-10-22_09.32.33.mp4

sfinexer, 2018-10-22:
@Harvie http://cnc.tiven.ru/ssr-2018-10-22_09.32.33.mp4
https://github.com/vlachoudis/bCNC/issues/1006

closed 2018-10-29 by Harvie