vlachoudis/bCNC #1014: new install after windows10 autoupdate

FXDUTRUCROSSET, 2018-10-20:
hello
after an autoupdate on win10 64 bits that completed apparently well but needs to reinstall windows keeping my files
i had to reinstall python 2.7 then pyserial and actual version of BCNC
but after many tries when i launch bcnc i just have a cms console black screen  with a few  lines of commands but bcnc that was working flawlessly on my laptop, no longer wants to works
does someone already occurs this situation and sorted it?
best regards

MARIOBASZ, 2018-10-20:
Check to install the version of @harvie or @sfinexer (problem with GIF / JPG files)

FXDUTRUCROSSET, 2018-10-21:
i ve tried on win 10 and winxps p3 and finally vista
i've sorted the case adding manually numpy-1.7.1-win32-superpack-python2.7.exe that was missing
and recalled for easyintall pyserial 
than used forked version of harvy because it wan't booting bcnc original version

FXDUTRUCROSSET, 2018-10-21:
for information i did the same way on win XP sp3 but pkg.resources was missing in Python27 folder 
in repertory C:\Python27\Lib\site-packages so i copied/pasted it from Python34 folder and it worked

Harvie, 2018-10-21:
yes. there's good chance that numpy will remain as dependency in future, since lots of features use it and there are lots of good things that numpy can do for bCNC in future. See #948 for details.

Harvie, 2018-10-29:
icon problem fixed, numpy remains dependency

closed 2018-10-29 by Harvie