vlachoudis/bCNC #1015: Problem with arc rounding G2 G3

MARIOBASZ, 2018-10-20:
When generating routes, I had a problem with arc rounding G2 G3. I have solved it increasing the radius a percentage 1/1001: multiply the theoretical radius by 1.001 to obtain the value of the radius to use ..
@Harvie, I have not verified if this problem is also present in your trochoid.

Harvie, 2018-10-21:
Not sure what do you mean by generating routes. How to reproduce this error?

MARIOBASZ, 2018-10-21:
I am using my own trochoid plugin, randomly, sometimes BCNC warned that the arc could not be generated. Itro round error that was presented, in straight paths to use, for example
block.append ("g1 X" + str (xx) ....
instead of
block.append (CNC.gline (xx, yy ...
if the path is a very small value instead of zero, the command is executed, for example, as g1 x 2.35e-12, and warns error.

Harvie, 2018-10-21:
Still don't understand what do you mean. You have problems with g-code containing numbers in "engineering notation"? Are you sure it's not caused by code of your plugin? I had this problem in the past, but i've fixed it. Do you generate g-code lines by yourself or you use `fromPath()` to generate it? This should not happen if you use `fromPath()` i've been testing it a lot.