vlachoudis/bCNC #1017: Cant self level

collaxd, 2018-10-24:
I'm not able to auto-level my machine
I noticed some problems
first the connect button is always red, apparently the machine is connected but the button turns red, the motors move normally
according to the movement of the machine always starts at the wrong point and the small red dot (zero) does not move
Thank you and sorry for the bad english
[image]
[image]

Harvie, 2018-10-29:
What was the last version of bCNC with autoleveling that worked for you?
Have you ever done autoleveling on your machine?

collaxd, 2018-10-29:
Never, it's my first time

collaxd, 2018-10-29:
[
[image]

](url)

OliverHaak, 2018-11-16:
I seem to have the same problem. bCNC 0.9.11 running under Linux Mint 18.1 worked for me beginning of this year, also after updating from GRBL 0.9 to 1.1f. Now after a couple of months I pulled that CNC machine from the shelf again to do some PCB milling and I have a lot of trouble. Meanwhile I upgraded to Linux Mint 18.3:

- Status is always "Not connected" althoug there is a connection established via USB
- No coordinates are being displayed in MPos or WPos
- Probing the Z-Axis for zeroing worked yesterday, today the Z-axis is moving in the wrong direction

Something is seriuosly broken. I already tried out bCNC 0.9.14 and updating all the python-modules on my system to the newest version --> no change

Harvie, 2018-11-20:
> Status is always "Not connected" 

This is certainly wrong... It never shows this to me when arduino is connected. Are you sure that you selected the right USB port with the right baudrate and settings?

OliverHaak, 2018-11-21:
Attached is a screenshot of bCNC on my laptop. You can see that there is an actual connection between the machine and bCNC (as the button for closing the connection is enabled) and I am also able to move the axes with bCNC. It would even accept and process G-Code. It would simply not display the coordinates (MPos and WPos). Maybe I mixed something up for probing the Z-axis lately but the effect being visible in the screenshot remains...
Any suggestions welcome!

[image]

Harvie, 2018-11-21:
BTW i have the same problem on Windows XP. bCNC always says "not connected" in red. But it kinda communicates with arduino. I am not sure, what is the problem.

Are there any other users who experience this?

ERCOman1, 2018-12-18:
I am having a similar problem in Windows 10 using the most recent version of bCNC and a Smoothie board.  Here is the error I get.

[image]

ERCOman1, 2018-12-18:
This is what the config file has in it....

[bCNC]
width = 900
height = 650
sash = 340
windowstate = normal

[Connection]
port = COM3
openserial = 1
controller = SMOOTHIE

which seems right....  Have you seen anything like this recently.  Funny thing is that it was working a while ago.  From a bit of research I am getting the feeling that I have to update the firmware on the smoothie board, but I'm not sure how to do it.......
http://smoothieware.org/bcnc
hopefully it's not the Smoothie board itself malfunctioning.

Harvie, 2019-01-03:
I really don't know how can i reproduce this bug on Linux and GRBL. Any ideas?

Harvie, 2019-01-03:
@ERCOman1 can you try to install different pyserial version? it's kinda mess and i don't know which one works the best...

ERCOman1, 2019-01-03:
Alright thanks.  In my opinion the Smoothieboard v1 is not relevant
anymore.  I just substituted it for an Arduino Uno.  bCNC works fine now.

Cheers,
Daniel

Harvie, 2019-01-04:
@collaxd You have GRBL 0.9, please upload latest GRBL1.1 to your arduino board, i think there is some problem with legacy GRBL 0.9 support right now and i don't have enough time to test it.

BTW probing setup is relatively complex. depends on what kind of probe you use, if it's NO or NC, if you have troubles with shielding, what probing algorithm (g-codes) do you use. Maybe try inverting probe pin in bCNC settings. etc. etc... For example if you try to probe, but probe is already active by the time you start probing, it will fail. It takes evening to learn...

@OliverHaak any news on this one?

OliverHaak, 2019-01-04:
@Harvie : ok, checked with the new PC under a fresh installed kubuntu 18.04.1 LTS. Installed versions are:
Python 2.7.15rc1
numpy==1.13.3
Pillow==5.1.0
pyserial==3.4
(tkinter is installed via the package manager)

Same behavior as before, bCNC connects to the CNC1610 with grbl1.1, big red sign says "Not connected", machine can be controlled manually and processes G-Code, does not show the MPos and WPos, though.

If you think its worthwhile, I can create an account for you on this machine so that you do some tests ...

Harvie, 2019-01-04:
@OliverHaak can you please try with pyserial==3.0.1 ?

I currently have it working like this:

```
Python 2.7.15
numpy==1.15.4
opencv-python==3.4.2.17
Pillow==5.3.0
pyopengltk==0.0.2
pyserial==3.0.1
```
BTW can you can you check if you have rw permissions to your serial port?

```
[harvie@anemophobia Temp]$ ls -la /dev/ttyUSB0
crw-rw---- 1 root uucp 188, 0  4. led 17.08 /dev/ttyUSB0
[harvie@anemophobia Temp]$ id
uid=1000(harvie) gid=100(users) skupiny=100(users),5(tty),14(uucp),81(dbus),90(network),91(video),92(audio),93(optical),95(storage),96(scanner),98(power),109(fuse),982(docker)
```

How do you execute bCNC? do you execute it like `python2 -m bCNC` ?
If it's not working please try `pip2 install bCNC`

OliverHaak, 2019-01-06:
Yes, I do have rw access to the tty:

> oliver@pandora2:\~$ ls -l /dev/ttyUSB0 
> crw-rw---- 1 root plugdev 188, 0 Jan  6 12:03 /dev/ttyUSB0
> oliver@pandora2:~$ id
> uid=1000(oliver) gid=1000(oliver)
> Gruppen=1000(oliver),4(adm),24(cdrom),27(sudo),30(dip),46(plugdev),116(lpadmin),126(sambashare)

With pyserial==3.0.1. the results are not any different. Screenshot attached. Funny thing is that the red button says "No connection" and when you click on it, the dialog window says "Connection established".

BTW, both ways executing bCNC work: running it via python2 -m bCNC and also starting the "bCNC"-script from the distribution directory.

Still, I can offer you to login in to my machine and see the problem onsite.

 
[image]

Harvie, 2019-01-06:
Maybe some scope related race condition. This already happend to me once. This gonna be hell to debug.

Can you try if this happens using GRBL-sim? Or at least using this fake-grbl script of mine:
https://github.com/vlachoudis/bCNC/blob/master/tests/fake-grbl.sh

roamerP, 2019-01-11:
I have the same problem on Windows 10 x64. bCNC always says "not connected" in red. But it communicates with arduino. I am not sure, what is the problem.

This is the error I get in command prompt:

Traceback (most recent call last):
  File "D:\bCNC-master\ToolsPage.py", line 973, in __init__
    exec("import %s"%(name))
  File "<string>", line 1, in <module>
  File "D:\bCNC-master\plugins\slicemesh.py", line 32, in <module>
    import stl #FIXME: write smaller STL parser
  File "D:\bCNC-master\lib\stl\__init__.py", line 1, in <module>
    from .stl import BUFFER_SIZE
  File "D:\bCNC-master\lib\stl\stl.py", line 12, in <module>
    from . import base
  File "D:\bCNC-master\lib\stl\base.py", line 21, in <module>
    class Dimension(enum.IntEnum):
AttributeError: 'module' object has no attribute 'IntEnum'

OliverHaak, 2019-01-11:
@Harvie : Sorry, I was unable to get GRBL-sim running. It seems very complex to get it compiled. Or is there a binary version somewhere?
Also no luck with your fake-grbl script, bCNC would just display an error-message window saying [Errno 22] Invalid argument.
Starting bCNC with taskset also showed no difference: I can connect to the machine via /dev/ttyUSB0, it accepts and processes G-Code, but no display of MPos and WPos. Shows always "Not connected"

Harvie, 2019-01-11:
> Sorry, I was unable to get GRBL-sim running. It seems very complex to get it compiled. 
@OliverHaak It's not really that hard.

```bash
git clone git@github.com:gnea/grbl.git
cd grbl/grbl
git clone git@github.com:grbl/grbl-sim.git
cd grbl-sim
make
./simport.sh
```

Then while the simport is running you connect bCNC to `/tmp/ttyFAKE`

Harvie, 2019-01-11:
> grbl-sim to or to run grbl on a generic arduino simulator

I've managed to get both to work. But there are bugs. Grbl-sim is alpha and grbl in simulator is not really happy. I've tried yesterday. There was strange behaviour in both. Some of it was caused by using latest GRBL master, which is buggy. But even after checking out stable tag it was not completely identical to GRBL running on arduino.

Harvie, 2019-02-05:
> Changing it back to the default ($10 = 1) did the trick.

Yes. we need better documentation for this. Maybe even automated check.

OliverHaak, 2019-02-08:
Sorry, took me a while to be able to test: Awesome, that also did it for me! Thanks to @Harvie and @RStaunlich for the efforts!
bCNC rocks!