vlachoudis/bCNC #1018: Unable to run bCNC on OSX

freddykat, 2018-10-27:
Hello, 
Been trying for past days to install bCNC on my OSX10.11
Im using python2.7 and installed the adicional packages as per your guide.
When i try to run the bCNC.py I got this error:

`Last login: Sat Oct 27 11:29:25 on ttys000

MBPdeFrederico:~ fredericooliveira$ cd /Users/fredericooliveira/Documents/BCNC/bcnc

MBPdeFrederico:bcnc fredericooliveira$ python bcnc.py

Traceback (most recent call last):
  File "bcnc.py", line 2592, in <module>
    application = Application(tk)
  File "bcnc.py", line 113, in __init__
    self.ribbon = Ribbon.TabRibbonFrame(self)
  File "/Users/fredericooliveira/Documents/BCNC/bCNC/Ribbon.py", line 494, in __init__
    image=Utils.icons["new"],
KeyError: 'new'

MBPdeFrederico:bcnc fredericooliveira$ `

Already tested the Tkinter and it works ok, but i dont seem to get why this happens.
Hope someone can help me out

walterwissmann, 2018-10-27:
The same issue as here ;-) :
https://github.com/vlachoudis/bCNC/issues/959

freddykat, 2018-10-28:
> It is a problem with the PNGs
> 
> The same issue as here ;-) :
> #959

Thanks walterwissman,
I've Succesful installed in win but it connects but wont say connected. lol

Harvie, 2018-10-29:
problem with icons fixed

closed 2018-10-29 by Harvie