vlachoudis/bCNC #1019: bCNC to MKS DLC 2.0 Connection Issue

colinluthier, 2018-10-28:
The MKS DLS uses Grbl 1.1. When I connect the Open button switches to Closed. The Banner reads "Not Connected" but when I click on th banner it reads "Connection is established with Grbl." I can't get any echo from the terminal. I am assuming there is a baud problem but I can't seem to get it sorted. Thank you for any ideas.

Harvie, 2018-10-29:
Do you have pyserial installed? what platform are you running bcnc on?

colinluthier, 2018-10-30:
It was running fine on a different port on my Arduino so I expect all dependencies are still correct. I am using a RPi 3. I had to switch ports to connect to the MKS.

Harvie, 2018-10-30:
So this only happens on that one particular board, while regular arduino with grbl works for you?

colinluthier, 2018-10-30:
Yes. That’s what makes me think I have a band rate wrong or something simple.

colinluthier, 2018-10-30:
Baud...

Harvie, 2018-10-30:
Can you connect to board using putty or screen? That should confirm you a baudrate.

colinluthier, 2018-11-11:
It took me awhile to get back to this. If you are using the MKS DLC 2.0 board go to the Arduino IDE and flash GRBL 1.1 onto it. Use the Arduino Uno setting. Then everything connects.

closed 2018-11-11 by colinluthier