vlachoudis/bCNC #1026: bCNC is not running

hapav, 2018-10-31:
Hello,

I have a problem with starting my bCNC. I install program how its on this video
Everything go fine until I start it. window of cmd and Python opens and half sec later close and nothing happen. I tryed to use the same versions of programs which re at the video and i also tryed to use latest versions. Still the same problem. 
Can you help me please?

Harvie, 2018-10-31:
Latest version of bCNC needs not only pyserial, but also numpy to be installed.﻿

closed 2018-10-31 by Harvie

hapav, 2018-10-31:
Thanks for that, but I'm maybe stupid. Can you tell me please what should i do to instal only pyserial and that start bCNC?

Harvie, 2018-10-31:
You can't run recent bCNC with only pyserial. You also need numpy.

Harvie, 2018-10-31:
There's all-in-one installer in progress, you can watch #437

hapav, 2018-10-31:
Thank you. It's running :)