vlachoudis/bCNC #1027: Compensate island radius is working?

lrp1985, 2018-11-02:
Hello, I load a dxf file with a circle with 10mm radius, then I select the code and toggle Island I check "Leave Island uncut" "Compensate islands for cutter radius". When toolpath is generated I don't see any difference or magin added equal to tool radius.

Do I doing something wrong?

Harvie, 2018-11-02:
"Island" is shape that is NOT cut. You select the shape and click on "island" it will then turn red.
If you create tabs, it's just special case of island. tabs are always compensated for cutter radius.

with:

[image]

[image]

without:

[image]

[image]

lrp1985, 2018-11-02:
So If I want just to cut a circle 20mm radius using an endmill of 3mm How I can specify the margin of the tool to get the real diameter and not a circle with 20mm - 3 mm = 17.

It can be done?

Harvie, 2018-11-02:
are you looking to cut that circle or make island?
if you are looking to cut that circle, you should use "profile" or "pocket" first to offset the endmill radius.
and then "cut" the resulting shape. before "cut" you also can add "tabs" on that shape.

lrp1985, 2018-11-02:
just to cut ok I got it