vlachoudis/bCNC #1031: Status bar not working when a code runs a second time

einencool, 2018-11-05:
(Copied from issue #885)

I've got one question, but I didn't know, if it is a new "issue" or not.
But with a Version from August I was working on till now, that issue wasn't there.

The first time I start a program everything is fine, but the next parts, the status bar won't work any more.

This is the status bar when the program is running the first time
[image]

after the first job, it shows me how long the program was running etc..
[image]

but when I start the next program, the status bar doesn't show anything
[image]
nothing about the buffer, or how long the program will run etc...

can you help me in this case?

This is a message when I close bCNC
[image]

Thank you in advance
Greets Chris

_Originally posted by @einencool in https://github.com/vlachoudis/bCNC/issues/885#issuecomment-435475329_

Harvie, 2018-11-06:
try now

einencool, 2018-11-07:
Hi @Harvie,

Tried today with Python 2.7.15 but this issue is still alive ...
[image]

Harvie, 2018-11-08:
Do you still have that "monitorserial" error message in cmd window?

einencool, 2018-11-08:
No, the cmd Box shows nothing like that. 

[image]

This was the cmd Box from yesterday. It was 3x the same small g-code

einencool, 2018-12-12:
So it seems to be an issue only happening when using the hardware buttons...

Harvie, 2018-12-13:
Lets wait until #1032 gets resolved. I think this might be already fixed once we figure out what's going on in #1032...

einencool, 2018-12-13:
I think you might be right 👍

closed 2020-01-09 by einencool