vlachoudis/bCNC #1032: Program crashes when starting with hardware "Start" button

einencool, 2018-11-05:
Hello @harvie,

a few days before you updated bCNC so that I can start the code with a click on the hardware "start" button.

Today I got some strange problems. I haven't change anything in the soft- or hardware, but today when I started a program the second time with the hardware button, the program chrashes without a message. Also python was completly closed. 
The first time I thought, I pressed something wrong, but also after a restart from my laptop the problem remains. 

The first time it starts correct, but the second run always crashes.
Since everything closes, I have no error message.
Can you please tell me, how I can help you finding the problem?

Greets
Chris

einencool, 2018-11-06:
Hi @harvie,

this only happens when I pressed the hardware button. With the software button or the "run" command on the external keypad everything was fine.

The first day I tried it, there was no Problem. It was the same G-Code, thats why I'm wondering, because I didn't update the program...

This is a short program, where the issue happens, but I have 3 files on that day, where the problems happen. 
[Acryl - 011 - Mittelloch - 3,175 Einschneider.txt](https://github.com/vlachoudis/bCNC/files/2554747/Acryl.-.011.-.Mittelloch.-.3.175.Einschneider.txt)

I will try the new bCNC Version tomorrow, today it's a bit late to make some noise :-)

Greets Chris

Harvie, 2018-11-06:
Firstly. These lines are not valid g-code:

```
( Fräser : 3000 = Schaft (3,175 mm) )
( Werkzeug:          Schaft (3,175 mm) )
```
Note the ) brace which prematurely closes the comment, which causes the last brace to cause problems. Some g-code interpreters (like CAMotics cnc simulator) fail on this. I had to remove it to simulate your code succesfully:

[image]

BTW you can generate much cleaner g-code using DXF drawing (or SVG or STL) of desired shape and PROFILE and CUT operations of bCNC (to make cuter radius offset and Z steps or spiral to cut down)

However bCNC can handle this problem with braces and for me it works flawlessly. I've tried two times using HW button, one time using SW and two more times using HW. No problem at all.

As i said: add pause to the end of .bat
Also you might have problems with your wiring. eg. electromagnetic noise from spindle crashing the arduino due to unshielded cabling. i had lots of troubles with this when i was beginning with grbl.

einencool, 2018-11-06:
Hi @Harvie,

thank you for your fast and helpfull reply.
I will delete the "(xxxmm) in the comment section in my software.
I work with Vectric and the PostP is not so good like the one in Fusion (adaptive clearing etc).
But for my Work it's easier to work with Vectric .

The G-Code is a few days old, because it was a little plate for my father is this has to be added later...
But I got some Probs with other older G-Codes too.
They were working without problems in the last days. (and with the "software" button they start properly).. 
I will check tomorrow with this files, and then I will give you a feedback :-)

The wiring for the stepper / switches etc is done with shielded cables, and the USB cable is shielded too.
When I build the machine, first there were no shielded cables, but then I got some "alarms" while the machine was running, that was not funny...

The problem yesterday was, that the spindle didn't start to spin, so it could be possible like you said, that the braces make the problems....

Greets Chris

Harvie, 2018-11-06:
> problem yesterday was, that the spindle didn't start to spin, so it could be possible like you said, that the braces make the problems....

For me spindle starts even with braces

einencool, 2018-11-06:
Now I come from my machine.

I've tested the new Version of bCNC, and I added the "pause" option at the end of the bcnc.bat
But I think I've done something wrong. the first 3 or 4 attempts start without a problem, but then the software crashes and closed completly, the only window that remains was this:
[image]

All other windows were closed.
Then I tried it another time, and there were no problems after 8 or 10 starts.
[image]

I opened and closed the software sometimes and I always make my homing Routine, but sometimes, the "Idle Status" was in Red and not the white colour...
I don't know why, and the G-Code starts absolutly normally...
[image]

einencool, 2018-11-06:
Now I removed the braces from the G-Code, and the program crashes 3 times again.
So that should not be the problem with it...

Harvie, 2018-11-06:
Idle status is always red for me. maybe it was not like that in older versions. but since i've started using bCNC (not that long ago) i remember only red idle.

This is very strange. Do you have latest python from 2.7 branch? I have 2.7.15

einencool, 2018-11-06:
I startet the program in a smaller window, and could see when something like "stream..." appiered in the command box, and then bcnc crashed again

I've got python 2.7.14
is there a simple way to update it?

Harvie, 2018-11-06:
> is there a simple way to update it?

dunno. i use linux. probably download .exe and click on it?

>  and could see when something like "stream..." appiered in the command box

yes it prints out stuff like this:

```
Stream requested by CYCLE START machine button
Ignoring machine stream request, because of state:  Idle True
Ignoring machine stream request, because of state:  Idle True
Ignoring machine stream request, because of state:  Idle True
Stream requested by CYCLE START machine button
Stream requested by CYCLE START machine button
```
it's only information that HW start was activated (or ignored, because it's already in progress).
did the cmd window stayed open after the crash?

einencool, 2018-11-06:
No, the comand windows crashes also. Only the window you can see a few answers ago stays open :-(

Harvie, 2018-11-06:
Do you have enough RAM?

Harvie, 2018-11-06:
try to replace

`start python "%DIR%bCNC.py"`

with

```
python "%DIR%bCNC.py"
pause
```

einencool, 2018-11-06:
I think 8gb should be enough :-)
It's a Core I 3 8xxx

Oh, after the new insallation of python I need to reinstall all the other stuff too?
Because bCNC doesn't start, and the cmd box closes ...

So I will look at it tommorrow. It's late and the night is short...
But thank you for your fast support, I enjoy it very much !!!

einencool, 2018-11-07:
So, today I got the program and the machine running again.

It seems to me, that Python 2.7.15 is the dealbreaker...
Now everything works fine.
I startet different codes several times without any issues (totally 15 times).
I also changed the bcnc.bat with the start command and now the cmd box stays open until iI press the key.

In this picture I startet a little code and it works fine
[image]

So for me it seems, that this issue has been solved :-)
Thank you for your premium support 🥇

Harvie, 2018-11-08:
You're welcome.

Good to know that. I wonder what the problem was with python 2.7.14...

einencool, 2018-11-10:
Hi @Harvie,

Today I‘ve made several pieces and one time the issue occurs, but I removed the „pause“ cmd in the .bat file. So there was no cmd box where I can see the error.

One question, is there the possibility that the switch is bouncing to much and that makes the problems? 
It was the only time I didn‘t press the switch „very hard“, but I wasn‘t able to reproduce the issue...

Greets Chris

Harvie, 2018-11-13:
Bouncing should not make any problems. Once the HW start is triggered any other starts are ignored.
Hence the `Ignoring machine stream request, because of state:  Idle True` message.

einencool, 2018-11-13:
Ah OK, 
I see what you mean :-)

I think tomorrow I‘ll have to make some parts and then I‘ll make some tests.
Yesterday it happend one time again, and foolishly I forgot to add the „pause“ cmd to the bat file.

After that I added it, but the issue didn‘t happen again...

I hope to give you a point where you can start your search :-)

einencool, 2018-11-15:
Hi @Harvie ,

Today I made a few pieces again.
And 2 times the program crashes after the second or third piece.

The cmd Box stays open and showed the following.
But it seems to me, that the "error message" appears earlier than the error...

[image]

Harvie, 2018-11-15:
Unfortunately this error message is not related to your problem. It's just that some file on your disk have name with funny characters in it and python2 can't handle it. This will get solved in python3 release of bCNC #236 #940 #1006 

But this should not cause crash of bCNC.

Harvie, 2018-11-23:
Please try latest version. I've fixed some weird bug that was happening only on certain systems. Not sure that it will help,  but at least give it a try.

einencool, 2018-12-07:
Hi @Harvie 

Today I made some pieces. The version I have installed is around 10 days old.

I made a piece and had to change a bit of the code. 
I changed it with fusion 360 and bcnc reloaded it on it’s own. That happens every time. So far so good, but then I started the code and not like the crash’s before only bcnc crash’s... this time bcnc crash’s and the program starts to run...
That would have been very dangerous if I hadn’t installed some hardware buttons to stop the machine...

Do you have an idea what was happening?

Harvie, 2018-12-08:
This is not enough info for me. Please try to get error messages.

einencool, 2018-12-09:
I know what you mean.
One question, you have changed the starting routine from bCNC in the bcnc.bat.

How can I change it, that the window will stay open, because it all chrashes...

start python -m bCNC
pause

doesn't work, 
can you please help me with it?
Greets Chris

Harvie, 2018-12-09:
remove the "start"

einencool, 2018-12-09:
oh, sometimes it can be so easy :-)

Harvie, 2018-12-09:
start causes program to be started detached from .bat window (so that it can close once bCNC is started)

einencool, 2018-12-09:
alright, thank you.
I will test it in the next days when I'll continue my work on some pieces.
wish you a nice weekend :-)

einencool, 2018-12-11:
Hi @Harvie 
This was the cmd box right now.
I made a piece, everything was fine, then I prepaired my machine for the next piece, loaded the G-Code and then by pressing the hardware button the Program crashes again. This time without starting the spindle and without running the Code.

`new-config Utils <ConfigParser.ConfigParser instance at 0x0000000002D7E908>
Controller state changed to: Alarm (Running: False)
Controller state changed to: Home (Running: False)
Controller state changed to: Idle (Running: False)
Controller state changed to: Run (Running: False)
Controller state changed to: Idle (Running: False)
D:\CNC\bCNC\bCNC\lib\bFileDialog.py:179: UnicodeWarning: Unicode equal comparison failed to convert both arguments to Unicode - interpreting them as being unequal
  if path not in _history:
Controller state changed to: Run (Running: False)
Controller state changed to: Idle (Running: False)
Controller state changed to: Run (Running: False)
Controller state changed to: Idle (Running: False)
Controller state changed to: Run (Running: False)
Controller state changed to: Idle (Running: False)
Controller state changed to: Run (Running: False)
Controller state changed to: Idle (Running: False)
Controller state changed to: Run (Running: False)
Controller state changed to: Idle (Running: False)
Controller state changed to: Run (Running: False)
Controller state changed to: Idle (Running: False)
Controller state changed to: Run (Running: False)
Controller state changed to: Idle (Running: False)
Controller state changed to: Run (Running: False)
Controller state changed to: Idle (Running: False)
Controller state changed to: Run (Running: False)
Controller state changed to: Idle (Running: False)
Controller state changed to: Run (Running: False)
Controller state changed to: Idle (Running: False)
Controller state changed to: Run (Running: False)
Controller state changed to: Idle (Running: False)
Stream requested by CYCLE START machine button
Will clean after this operation
Controller state changed to: Run (Running: False)
Controller state changed to: Idle (Running: False)
Job done. Purging the controller. (Running: False)
Controller state changed to: Run (Running: False)
Controller state changed to: Idle (Running: False)
Controller state changed to: Run (Running: False)
Controller state changed to: Idle (Running: False)
Controller state changed to: Run (Running: False)
Controller state changed to: Idle (Running: False)
Controller state changed to: Run (Running: False)
Controller state changed to: Idle (Running: False)
Controller state changed to: Run (Running: False)
Controller state changed to: Idle (Running: False)
Controller state changed to: Run (Running: False)
Controller state changed to: Idle (Running: False)
Controller state changed to: Run (Running: False)
Controller state changed to: Idle (Running: False)
Controller state changed to: Run (Running: False)
Controller state changed to: Idle (Running: False)
Controller state changed to: Run (Running: False)
Controller state changed to: Idle (Running: False)
Controller state changed to: Run (Running: False)
Controller state changed to: Idle (Running: False)
Controller state changed to: Run (Running: False)
Controller state changed to: Idle (Running: False)
Controller state changed to: Run (Running: False)
Controller state changed to: Idle (Running: False)
Controller state changed to: Run (Running: False)
Controller state changed to: Idle (Running: False)
Controller state changed to: Run (Running: False)
Controller state changed to: Idle (Running: False)
Controller state changed to: Run (Running: False)
Controller state changed to: Idle (Running: False)
Stream requested by CYCLE START machine button
Will clean after this operation
Drücken Sie eine beliebige Taste . . .`

einencool, 2018-12-12:
Today I made some pieces with the machine, and the program crashes several times.
Then I switched back to the software buttons and everything worked fine...

When I was running the code after I started it with the software buttons i saw that the status bar was working correctly. And the code that was done was shown in green on the screen. All that doesn’t happen when you start the code with the hardware buttons.

Today I paused the code by hardware button and then I pressed start to continue the code, but then the software runs only for a few seconds and then the machine stops but kept the spindle running. No errors or something like that..

So I’ll switch back to the software buttons because of the big problems with the crashes...

Harvie, 2018-12-13:
>  Then I switched back to the software buttons and everything worked fine...

At least it's not broken completely. I think, we might investigate the difference between software and hardware button handlers.

Any idea, why can python suddenly shutdown without printing any error message??? I am python newbie and i don't know how to debug such case.

einencool, 2018-12-13:
Sadly I’ll be no help for you.
I don’t understand anything from coding...

Harvie, 2019-12-01:
Can you please try with latest git master version?

einencool, 2019-12-01:
In the last months i didn‘t do so much with my cnc, and when I used it, I was only working with the software buttons. I hope in the next two weeks I'll give it a try and give you a feedback.

Harvie, 2019-12-02:
Haha. Still better than me. I've buyed these buttons like year ago and they are still waiting in drawer at my office to be installed to the machine :-) I started to implement the software side first, but then forgot to do the wiring :-)

einencool, 2019-12-13:
So, yesterday and today I’ve made several pieces of wood on my machine and for now it works absolutely perfect ☺️
Now I’m at piece 40 and there are 120 more to go, so I’ll give you a feedback in the evening when I hope everything is ready ...

Thank you for your continuous support, that’s the best software for me

einencool, 2020-01-09:
It's a little bit later, but here is my response about the changes :-)
It works really good, but sometimes the programm crashes while running the g-code.
It was not the behavior that was otherwise the case.
When the programm crashes, the hardware and software buttons didn't work anymore.
The problem is, that I saved the log files, but I can't find them anymore because my network drive had a fault. So the only point I have is a screenshot that I'd saved on the desktop
Maybe it can help you a little bit
[image]

einencool, 2020-01-09:
Well, I've found the missing Error Logs :-)
I hope they will help you a little bit

[bcnc Fehler 1 - 13.12.19.txt](https://github.com/vlachoudis/bCNC/files/4042210/bcnc.Fehler.1.-.13.12.19.txt)
[bcnc Fehler 2 - 13.12.19.txt](https://github.com/vlachoudis/bCNC/files/4042211/bcnc.Fehler.2.-.13.12.19.txt)