vlachoudis/bCNC #1033: Unable to see "Config", "Controller", "Shortcut", etc. under the Tools tab

Lester444, 2018-11-05:
After loading a newer version of bCNC with the hardware "Run" button ........... 

@Harvie , dunno why but I can no longer see "Config", "Controller", "Shortcut", etc. These used to be under the "Tools" tab.

Is it because I am using "Double Size Icon"? (I need big icons for my touchscreen monitor...... & my stubby fingers)

I had to go back to a previous version of bCNC which I have been using June 2018 (sorry but I can't see the bCNC version number on my screen).  I hope you can fix this since the hardware "Run" button is working great for me.
 
reference: Issue #885

Lester444, 2018-11-06:
I am not able to disable the "double size icon"  because I can't see "Config".

Harvie, 2018-11-06:
You can disable that in bCNC.ini file

Harvie, 2018-11-06:
However for me it works in both modes. Are you sure that you have latest bcnc master from git, and you run python2 with numpy module installed?

[image]

[image]

Harvie, 2018-11-06:
I think i know what is going on. I guess that your screen is too smal... Right?

Lester444, 2018-11-06:
I am using 1024 x 768 resolution.

Here are some pics:
[image]
[image]

I'd rather stick with the double-sized icons to minimize inadvertent mis-pressed icons.  Just trying to avoid a crash. I am using a touchscreen monitor.

Harvie, 2018-11-07:
This is not really bug. Your screen is just too small :-)
I don't know how to handle this. If i optimize to 1024px There can always be someone with smaller screen who will start complaining. Maybe we can add some kind of scrolling mechanism to ribbon.

But to be honest i don't think that these buttons you have troubles with should be in "tools" tab. I think i will move it to "file" or "control", which is less populated, so that should solve your problem.

Lester444, 2018-11-07:
I guess I will have to stick to the older version then. It is simple, less bells and whistles, more robust.

Having a reliable g-code sender is much more important to me than one with more features. So far I have machined more than 2,500 individual pieces on my 2x CNC machines and I think that speaks a lot for the robustness of this combination (GRBL & bCNC). And the low resolution display is great for touchscreen applications.... and feeble eyesights.

Thank you @Harvie for your efforts, much appreciated.

@sfinexer, I don't use a keyboard on my CNC machines, so I don't have access to Alt key. The touchscreen serves my needs 99.9% of the time.

Lester444, 2018-11-07:
>  If i optimize to 1024px There can always be someone with smaller screen who will start complaining.

Look at it from a different light: If you keep on adding buttons then sometime in the future a lot of people will complain.  Not all users will have the same resolution as what you have.

Harvie, 2018-11-07:
Maybe in meantime you can add some page and photos to wiki about hardware you use in such setup. That might be interresting.

Lester444, 2018-11-07:
>  cheap raspi lcd

I have this display but gave up on it after 2 days of use. The bCNC icons are terribly small even on double-size, necessitating the use of a stylus pen...... but which didn't work at all because the screen is capacitive-type (it needs some surface area in close proximity to the spot on the screen). So with oily fingers (cutting fluid) trying to press a tiny spot....... that took a lot of attempts and inadvertent runs (the shortcuts I made) before finally hitting the right icon. Very frustrating.

What I'm using now is 15" and it fits the need.

Harvie, 2018-11-08:
I leave this open, so i know i should look at this before releasing next version.

Harvie, 2019-01-08:
Can you please test the latest git version? @vlachoudis did some great work and UI should be more compact now!

Lester444, 2019-01-08:
Acknowledged, @Harvie. Will try it ASAP and report my findings here.

Lester444, 2019-01-09:
Tried it. Now the "Tools" tab is gone. Can't use this version.  :'(

Harvie, 2019-01-09:
"Tools" tab is not gone. It was renamed to "CAM" and stuff that wasn't CAM related was moved to another tabs. We did this in order to fit into your little screen.

Harvie, 2019-01-09:
Also note, that currently there is bug. so if you click on any of these buttons:
[image]

you have then to switch to "CAM" tab. we will fix this before release.

Lester444, 2019-01-09:
OK I see it....... now under "FILE" tab. So I switched off double size icon, exited from bCNC, re-launched bCNC, but now the "Config", "Controller" buttons do not work.

Harvie, 2019-01-09:
As i said... you have to click "config" and then immediately switch to "CAM" tab. it's a bug.

Lester444, 2019-01-10:
I don't understand what you wanted me to test....... since you already know there is a bug. Anyway, there's your confirmation that indeed there is a bug.

> Can you please test the latest git version? @vlachoudis did some great work and UI should be more compact now!

vlachoudis, 2019-01-10:
I've restored the Config inside the "CAM". Unfortunately is not straight forward to move it
to the more logical place of "File" tab

sfinexer, 2019-03-11:
Well, there is simply no logic.
Tab named as:
```
	__doc__ = _("GCode manipulation tools and user plugins")
	_name_  = N_("CAM")
```

Therefore, it is necessary to remove the controller parameters from it, you need to create a new tab, and there ONLY settings should be placed.

Harvie, 2019-03-11:
> Therefore, it is necessary to remove the controller parameters from it, you need to create a new tab, and there ONLY settings should be placed.

Thanks for clarification. PR welcome!

sfinexer, 2019-03-13:
> > Therefore, it is necessary to remove the controller parameters from it, you need to create a new tab, and there ONLY settings should be placed.
> 
> Thanks for clarification. PR welcome!

Because it is necessary to use "rapid application development". And do ALL visual manipulations using the CURSOR, and not write something there in a text file. Now 21st century.

Maybe someone wants to help, but cannot (((You have to rewrite everything using wxWidgets, for example, if you need to cross-platform.

After all, this is a GUI and not a console program for various calculations.

Harvie, 2019-03-13:
> You have to rewrite everything using wxWidgets, for example, if you need to cross-platform.

No need to discuss this here, there are already other issues on this topic.
I've even tried to build new UI using GTK3 builder. I think it's nice solution, because you have separate UI definition in XML file and you have UI independent code in python. So if you want to change the UI, you just modify the XML without having to modify python (there's "Glade" software do modify it visualy by drag and drop).

wilderf353, 2019-03-18:
Is there a desire to go to wxPython?

Harvie, 2019-03-18:
> Is there a desire to go to wxPython?

No. Why? I think GTK3 builder is much better concept.

lyatesbme, 2019-08-29:
HI Harvie
Thanks for the include to this thread. I thought that others would have experienced this issue. I noticed on the Wiki for bCNC a Video from a daSamo123 on Youtube that shows a 5 inch LCD display with the bCNC fitted and with a right side and bottom side scroll bar. Was this an earlier version or have I missed something in my setup.

Many Thanks

art-x, 2020-04-06:
I needed to have bCNC on a 3.5 inch RPI display (on my Raspberry Pi 3a) with touchscreen. After alot of tweaking in the GUI I managed to get all main components within the confines of 480 x 320 (landscape) TFT.
A lot of the configuration worked out as expected, except for the DRO and Ribbons (I wanted the icons without text as much as possible due to space restrictions).
It required me to change some of the source code to make things fit.
When started with -f (fullscreen), all common functionality is accessible.

This is the final result: (3.5 inch 480 x 320 TFT RPI touch screen)
[image]
[image]
[image]

If anyone is interested, I'll commit the changes I made. 
Harvie, it is crude code for now, but I feel it might be feasable to incorporate this in the config?

Harvie, 2020-04-06:
> Harvie, it is crude code for now, but I feel it might be feasable to incorporate this in the config?

But that would limit functionality for users with regular sized monitors right?

And the config/controller buttons are still unavailable. But i have some ideas how to improve this situation... I will commit it later...

Harvie, 2020-04-06:
I've fixed the original problem of @Lester444 so i closed this issue.
@art-x if you have some other problem, please open separate issue.

closed 2020-04-06 by Harvie

BjarneJensen, 2020-04-28:
@art-x I am trying to do the same as you but cannot seem to find the right settings to edit. I changed some font settings and tried looking at the control page code but could not figure it out. Sorry I know this is closed but hopefully it was recent enough that you remember. Thanks!

RemyFr, 2020-11-20:
> I needed to have bCNC on a 3.5 inch RPI display (on my Raspberry Pi 3a) with touchscreen. After alot of tweaking in the GUI I managed to get all main components within the confines of 480 x 320 [...]

Hello art-x,

Your configuration is exactlly what i am looking for my home made CNC, could you give us your config/protocol to have this result ?

You will save me, it is so frustrating to be blocked in this step.

Remy

kd6vpe, 2021-01-01:
Wish this setup could be shown. It is what I could use also. At the moment I just use a wireless keyboard and the arrow keys to control my jogging. Not able to control the movement distance this way though like setting to .1 inches.

intelka-net, 2023-07-02:
> 

Hi! Could you please share your config file? Or where exactly did you change the layout. Thank you

commit 5674462b (2020-04-06): Moved config to lefmost side of ribbon. Fixed #1033