vlachoudis/bCNC #1035: Stop during operation.

sfinexer, 2018-11-08:
Stop during operation.
And the Z coordinate is not displayed correctly.

Harvie, 2018-11-08:
Not enough informations

sfinexer, 2018-11-08:
[drawings-of-tigers-tiger-drawings-related-keywords-amp-suggestions-tiger-drawings(270мм X 210мм).cncR.zip](https://github.com/vlachoudis/bCNC/files/2560991/drawings-of-tigers-tiger-drawings-related-keywords-amp-suggestions-tiger-drawings.270.X.210.cncR.zip)
and press stop in the process of the machine.

Harvie, 2018-11-08:
1.) Do you use latest GRBL version?
2.) What WPOS coordinates does GRBL report when this happens?
3.) Do you use any offsets, WCS or similar stuff in your setup or gcode?
4.) What safe-z you have set?

Harvie, 2018-11-08:
WTF are these?

[image]

sfinexer, 2018-11-08:
1) grbl_1.1g
2) I press the stop button at any time.
3) 

> [Controller]
grbl_0 = 10
grbl_1 = 25
grbl_2 = 0
grbl_3 = 0
grbl_4 = 0
grbl_5 = 0
grbl_6 = 0
grbl_10 = 3
grbl_11 = 0.0
grbl_12 = 0.002
grbl_13 = 0
grbl_20 = 1
grbl_21 = 0
grbl_22 = 1
grbl_23 = 3
grbl_24 = 100.0
grbl_25 = 500.0
grbl_26 = 25
grbl_27 = 20.0
grbl_30 = 50000.0
grbl_31 = 2000.0
grbl_32 = 0
grbl_100 = 80.0
grbl_101 = 80.0
grbl_102 = 40.0
grbl_110 = 10000.0
grbl_111 = 4500.0
grbl_112 = 10000.0
grbl_120 = 3000.0
grbl_121 = 200.0
grbl_122 = 1000.0
grbl_130 = 319.0
grbl_131 = 429.0
grbl_132 = 300.0

4),,,,

@Harvie CP1251

Harvie, 2018-11-08:
> CP1251

I don't care. still not valid g-code

```
[Блок №91301]   
G0 Y267.127 X46.646
G1Z-5
G0Z1
[Блок №91302]   
G0 Y267.059 X47.965
G1Z-5
G0Z1
[Блок №91303]   
G0 Y267.092 X48.878
G1Z-5
G0Z1
[Блок №91304]   
G0 Y266.925 X49.353
G1Z-5
G0Z1
[Блок №91305]   
G0 Y267.086 X50.057
```

sfinexer, 2018-11-08:
@Harvie Why?
Where did you get that code is not correct?

Harvie, 2018-11-08:
As i said. AFAIK you can't use square brackets.

sfinexer, 2018-11-08:
I'll try the ")" brackets, I don't think something will change.

sfinexer, 2018-11-08:
@Harvie But the code works with these brackets!
Errors when performing no.

sfinexer, 2018-11-08:
@Harvie If I press the pause, then everything is fine, the error appears only when the stop button is pressed.

MARIOBASZ, 2018-11-09:
In the first video gijate that briefly appears error 22: @Harvie, I ask again, can you correct the removal of pop-ups for errors, which was enabled in old versions?
Must be able to read the error in "terminal".
Check online 5,
  G1 Z-5
It has no value of F.

sfinexer, 2018-11-09:
@MARIOBASZ The value of F is set once and remains unchanged until another value is assigned.

F- working feed rate, it does not depend on G0.
@MARIOBASZ In the second video, I initially paused, all the coordinates are correct, then I pressed the stop button, and the coordinates changed !!!

MARIOBASZ, 2018-11-09:
In terminal you read the error code? In your example I did not see the value of F before line 5

sfinexer, 2018-11-09:
@MARIOBASZ line 2.
No, I did not read the error description.

sfinexer, 2018-11-09:
@Harvie,
I need to skip the line!
If I use "()" instead of "[ ]" then grbl will look for the completion of a comment.
It is not advisable to use "()" in this case, below is the grbl code responsible for this.
In other words, I will unnecessarily load the arduino.

>   for (;;) {

    // Process one line of incoming serial data, as the data becomes available. Performs an
    // initial filtering by removing spaces and comments and capitalizing all letters.
    while((c = serial_read()) != SERIAL_NO_DATA) {
      if ((c == '\n') || (c == '\r')) { // End of line reached

        protocol_execute_realtime(); // Runtime command check point.
        if (sys.abort) { return; } // Bail to calling function upon system abort

        line[char_counter] = 0; // Set string termination character.
        #ifdef REPORT_ECHO_LINE_RECEIVED
          report_echo_line_received(line);
        #endif

        // Direct and execute one line of formatted input, and report status of execution.
        if (line_flags & LINE_FLAG_OVERFLOW) {
          // Report line overflow error.
          report_status_message(STATUS_OVERFLOW);
        } else if (line[0] == 0) {
          // Empty or comment line. For syncing purposes.
          report_status_message(STATUS_OK);
        } else if (line[0] == '$') {
          // Grbl '$' system command
          report_status_message(system_execute_line(line));
        } else if (sys.state & (STATE_ALARM | STATE_JOG)) {
          // Everything else is gcode. Block if in alarm or jog mode.
          report_status_message(STATUS_SYSTEM_GC_LOCK);
        } else {
          // Parse and execute g-code block.
          report_status_message(gc_execute_line(line));
        }

        // Reset tracking data for next line.
        line_flags = 0;
        char_counter = 0;

      } else {

        if (line_flags) {
          // Throw away all (except EOL) comment characters and overflow characters.
          if (c == ')') {
            // End of '()' comment. Resume line allowed.
            if (line_flags & LINE_FLAG_COMMENT_PARENTHESES) { line_flags &= ~(LINE_FLAG_COMMENT_PARENTHESES); }
          }
        } else {
          if (c <= ' ') {
            // Throw away whitepace and control characters
          } else if (c == '/') {
            // Block delete NOT SUPPORTED. Ignore character.
            // NOTE: If supported, would simply need to check the system if block delete is enabled.
          } else if (c == '(') {
            // Enable comments flag and ignore all characters until ')' or EOL.
            // NOTE: This doesn't follow the NIST definition exactly, but is good enough for now.
            // In the future, we could simply remove the items within the comments, but retain the
            // comment control characters, so that the g-code parser can error-check it.
            line_flags |= LINE_FLAG_COMMENT_PARENTHESES;
          } else if (c == ';') {
            // NOTE: ';' comment to EOL is a LinuxCNC definition. Not NIST.
            line_flags |= LINE_FLAG_COMMENT_SEMICOLON;
          // TODO: Install '%' feature
          // } else if (c == '%') {
            // Program start-end percent sign NOT SUPPORTED.
            // NOTE: This maybe installed to tell Grbl when a program is running vs manual input,
            // where, during a program, the system auto-cycle start will continue to execute
            // everything until the next '%' sign. This will help fix resuming issues with certain
            // functions that empty the planner buffer to execute its task on-time.
          } else if (char_counter >= (LINE_BUFFER_SIZE-1)) {
            // Detect line buffer overflow and set flag.
            line_flags |= LINE_FLAG_OVERFLOW;
          } else if (c >= 'a' && c <= 'z') { // Upcase lowercase
            line[char_counter++] = c-'a'+'A';
          } else {
            line[char_counter++] = c;
          }
        }

      }
    }

    // If there are no more characters in the serial read buffer to be processed and executed,
    // this indicates that g-code streaming has either filled the planner buffer or has
    // completed. In either case, auto-cycle start, if enabled, any queued moves.
    protocol_auto_cycle_start();

    protocol_execute_realtime();  // Runtime command check point.
    if (sys.abort) { return; } // Bail to main() program loop to reset system.
  }

sfinexer, 2018-11-09:
My code does not give errors.
[image]

Harvie, 2018-11-09:
bCNC cannot load your g-code file.
`UnicodeDecodeError: 'ascii' codec can't decode byte 0xcf in position 4: ordinal not in range(128)`

there is no way to choose codepage in bCNC. i can manualy convert it using iconv.

> If I use "()" instead of "[ ]" then grbl will look for the completion of a comment.
> It is not advisable to use "()" in this case, below is the grbl code responsible for this.

where did you find info, that you can use square brackets. in code you have posted is NO code that would handle square brackets. i think you are just wrong.

sfinexer, 2018-11-09:
@Harvie Maybe I'm wrong.
Here is the video, the code is loading, b-cnc is the latest version.
http://cnc.tiven.ru/ssr-2018-11-09_19.03.43.mp4

sfinexer, 2018-11-09:
@Harvie I will not close the topic, probably I will be able to understand what the problem is.

Harvie, 2018-11-09:
Does the fixed code i've attached work for you?

sfinexer, 2018-11-10:
@Harvie Yes, the problem is solved.
So the fault is not correct brackets.

sfinexer, 2018-11-10:
Error remained ((((

sfinexer, 2018-11-10:
@Harvie !!!!!!!!!!!!!!!!!!!!!!!!!!
http://cnc.tiven.ru/ssr-2018-11-10_20.41.44.mp4

sfinexer, 2018-11-10:
@Harvie https://github.com/grbl/grbl/issues/1460
This branch no longer develops, it says below:
https://github.com/grbl/grbl
"Grbl v1.1 has been released here!"
https://github.com/gnea/grbl

sfinexer, 2018-11-11:
@Harvie 
I think b-CNC mistake is that before executing your code I execute the above command G92.
I need it as this positioning axis position Z.
And pressing the stop button executes the G90 command.

But I did not ask to execute the G90 command. I press the stop button, this is a b-CNC error.

Harvie, 2018-11-12:
I've just realized one thing. The [square brackets] are bCNC syntax for variables... eg.:

```
G90
G0 X[xmin] Y[ymin]
G0 X[xmax]
G0 Y[ymax]
G0 X[xmin]
G0 Y[ymin]
```

Harvie, 2018-11-21:
OK. I am able to reproduce this: https://github.com/vlachoudis/bCNC/issues/1035#issuecomment-437601975

I will take a look at it...

Harvie, 2018-11-21:
If you just want to zero your WCS you should not use `G92`

Proper way to zero `G54` WCS is to use `G10L20P1X0Y0Z0`.
See http://wiki.linuxcnc.org/cgi-bin/wiki.pl?CoordinateSystems#G5x_work_coordinate_systems

You have to first read how g-code works.

Harvie, 2018-11-21:
And if you ever use `G92` (which is probably wrong in your use case). Then you should reset it afterwards using `G92.1` or `G92.2` you are using  `G92 X0.00 Y0.00 Z.00`, that will not reset G92!

closed 2018-11-21 by Harvie

Harvie, 2018-11-21:
In fact there was a bug, that bCNC didn't purged the controller after running job. It is instead purged when you press STOP button. bCNC should press the STOP button automaticaly after the job ends.
So that means your g-code will still end up in Z15 position.

I think i've fixed this problem, but it will not affect the user configurable buttons. Because i am not sure if it's good idea to purge after every button press. It will only affect g-code ran from g-code files.

You still have to learn how g-code works and fix your file.