vlachoudis/bCNC #1039: Correct Z machine Safe button

MARIOBASZ, 2018-11-09:
Good improvement to add the buttons.
In Z machine Safe eliminate line G54, it is dangerous!
Without that line it remains in current W Pos
uggested improvement:
To be able to visualize the current Parameters of End mil, Z safe, surface, etc

Harvie, 2018-11-09:
Currently we have this:

```
name.2 = Safe Z
icon.2 = rapid
tooltip.2 = Go to Safe Z
command.2 = G90G0Z[safe]
```

If i add G54 it might be dangerous, because you might not have G54 properly zeroed when you work in different offset.

MARIOBASZ, 2018-11-09:
I have downloaded the new version
Loaded with Z Safe button:
G53 g0 Z-2
G54
we talk about the same thing: Delete line G54

Harvie, 2018-11-09:
I have no G54

[image]

Harvie, 2018-11-09:
Maybe it's just in your bCNC.ini

MARIOBASZ, 2018-11-09:
Thanks. 
Also look at the go to z safe machine button. Hugs.

Harvie, 2018-11-09:
don't know what do you mean.

MARIOBASZ, 2018-11-09:
This button
[image]

closed 2018-11-09 by MARIOBASZ