vlachoudis/bCNC #1042: B-cnc instead of a comment displays empty lines.

sfinexer, 2018-11-15:
B-cnc instead of a comment displays empty lines.
[Narisovannyj-glaz V2.cnc.zip](https://github.com/vlachoudis/bCNC/files/2583693/Narisovannyj-glaz.V2.cnc.zip)
[image]

Harvie, 2018-11-15:
Comments work OK. Problem is with your alphabet. Python 2 cannot support this. I've already told you we know about this in #1006 (also i already have issue #940 on this topic).

No matter how many issues you create it will not change until we move to Python 3.

If you post any issue here, **please do not send g-code containing non-latin letters**. First test on english-only g-code. Otherwise you are wasting my time. The time that could be used to implement python 3 (and therefore cyrilic support) instead.

closed 2018-11-15 by Harvie

Harvie, 2019-11-28:
National alphabets and python3 support should be fixed in latest git master (not in pip release yet). Can you please try?