vlachoudis/bCNC #1044: Tangential knife, tangential control.

FatNarval, 2018-11-16:
Hello, it would be cool if  bcnc could have tangential knife option.

 I think it could be done just on bcnc side, no need to change grbl stock firmware.
Z axis pin could be used for rotational/ tangential axis, and z axis Up and down could be actuated using gas piston or solenoid etc, controlled via spindle on/off pin.

Some critical areas, except rotation of knife (tangential knife) there shold be tool lift control, when corner is to sharp, tool lifts rotates to new angle plunges down and starts cutting again. Controlling dealy between segments (lifting plunging) is also important for proper and efficient work.

What do you think?

Best regards!

Harvie, 2018-11-19:
That is possible, but i don't have time to do it. Are you willing to implement it? PR welcome :-) Just note it will need to add support for 4th axis to both bCNC and Arduino or whatever motion controller you use. current GRBL can't do this. Some others can.

Harvie, 2018-11-19:
Also have you tried the drag knife feature? It works very well once you get the hardware setup right.

colinluthier, 2018-11-20:
Does bCNC accept the drag knife offset and correct accute inside and outside cuts? I have seen code that loops out side cuts and picks up the knife and sets it down in scrap material to change the knife direction then restart the cut for inside angles.

FatNarval, 2018-11-21:
@Harvie 
Hello,and  sorry for late response.
I have just installeda the latest bcnc.exe it was a breeze! Great work! Thanks! 
I will test drag knife feature of bcnc. Since my drag knife has rather small offset, maybe I will try on vinyl film with some detailed geometry.
Also I will try drag knife with rotational blade (basicly swivel wheel)

Not sure i understand all terminology and protocols on github, but of course I would like to physicly implement it and test it.

I understand that 4th axis support one way to go, but what I had in mind is to totally different route. 
Idea is to totally avoid grbl changes, to use existing, stock grbl. Grbl still would work (compute) in just 3axis.
XY movement would be the same, Z axis would be totaly ignored (in up/down sense) and recalculated as rotational 4th axis, all computing will be done just on bcnc side. 

 Physical up and down movement will be actuated by on/off (full up /full down) control.  logical solution would be using spinndle on/off pin. Which is already implemented in grbl via M3 M5 command.
 Physicaly z axis would just go full or up full down.
Best way to achive this is pneumatic cylinder. It can be controled just by on/off , it is fast, it allows adjusting pressure/cutting force, it conforms to uneven surface, height differrnce, while reataing constang force (whitin cylinder travel).
Another way which can make fine(graduated)  physical controll of Z height would be to use S command, and spindle pwm output to control  an analog servo (cheap RC servo  with potentiometer as a feedback, or simmilar diy contraption).  It allows to use spring as cutting force source, and with depth of Z force could be moderated.
 I hope I have made things more clear than foggy with these explenations.
Best regards!
Keep in touch!

CCERocks, 2019-10-03:
Hi @FatNarval, what is your result on using drag knife gcode with rotary blade ?

FatNarval, 2019-10-12:
Hi, I havent tested it yet, I got all hardware but never got time to mount
it or play with it, sorry.
I was using vynil cutter knifes and it works great on coated fabric up to
~1mm of thickness like pvc, tpu tarps , and blade offset is marginal from ~
0,3 to  ~0,6mm so i just use normal toolpath.
My perdiction/expectations about rotary-drag are it won't be good for
detailed work but it adds functionality for cutting regular wowen or
delicate fabric which are impossible to cut by regular drag knife.