vlachoudis/bCNC #1045: Quilting, embroidery feature of bcnc

FatNarval, 2018-11-16:
Hello,  I think that bcnc has natural talent for controlling quilting embroidery machine!
Driller feature of bcnc looks so simmiliar to professional quilting or embroidery machines/software where needle position and needle piercing is  predetermined. 
Xy of grbl control xy of sewing frame, z of grbl  controls sewing machine pulley.
Some minor obstacles i see are:
Sewing machine rotates always in same direction, that could be solved in small change in firmware  or to keep grbl firmware original, dir pin can be physicly pulled to fixed state (ie. some other pin like gnd or +5v)
Z up an down travel combined should be equal to exactly one revolution of sewing machine pulley.
Safe Z  should be height of needle outside material, and is only allowed to move xy frame when needle is above safeZ, it allows sefly and acuratly sewing while keeping maximum efiviency, g0 movements should be only allowed during needle in top position, Z0.

Maybe most of this could be achieved with post processors, and existing bcnc software.

Any thoughts?

Best regards!

MARIOBASZ, 2018-11-16:
Brilliant idea!

Harvie, 2018-11-18:
Yes. Why not? But i don't have any experience with such machines to develop such features.
Maybe you can use POCKET to fill area and DRILLER to embroid that area.

> Safe Z should be height of needle outside material, and is only allowed to move xy frame when needle is above safeZ, g0 movements should be only allowed during needle in top position, Z0.

I guess this should be handled in motion controller...

FatNarval, 2018-11-21:
> Maybe you can use POCKET to fill area and DRILLER to embroid that area.

Exactly!

> I guess this should be handled in motion controller
I was thinking from general cam software perspective where this is handled by postprocessors.

FatNarval, 2018-12-01:
One important feature of (cnc )sewing machines is that at tht start and the end of each srgment  stiches go back and forth couple of stitches to lock the thread from unraveling.
In driller it could be implemented in thus way: it cycles for couple of times  2  or 3 of first/last holes (stitches) at start  or end of each segment.

wilderf353, 2019-03-18:
I have ordered parts to build a embroidery machine. It will be using a CNC shield on an arduino. I have not figured out to control it yet, but maybe I can figure out how to add it to bCNC.

(I have just started playing with GRBL & bCNC. I am helping my son and a few classmates with their senior project )

MARIOBASZ, 2019-03-18:
analogously to trochoidal paths, or driller plugin, you can generate the sequence back x steps, up, down, etc, and then go to the next step

FatNarval, 2019-03-18:
Simplest way is to use driller plugin to generate embroidery/ quilting path.
X and Y axses are used as normal.

Z axis stepper should be drive motor of sewing machine.
Step motor should be coupled to sewing machine pulley whitout slipping. Eg.
splined coupling etc. It realy helps when motor is in direct drive meaning
ratio is 1:1.

Dir pin of z stepper driver should be hardwired so the motor always turn in
same direction.
9
Also very important thing is timing and sync.
When tool is in safe z or top position neddle also should be in tdc.
When drill reaches bottom needle should also be in tdc.
How to achieve this?
There are two places where this parameter could be met. Grbl setiing steps
per mm for Z axis. And in driller plugin itself.
I reccomend using grbl setting just to set up resolution or virtaul
gearing, and to perform actual  "calibration" in driller plugin.
Eg. If you set in grbl 200 steps per mm, asuming no microstepping, in
driller safe z and drill depth combined should be equal to 1mm. It helps to
make safe z and drill depth same value in this hypothetical case 0,5mm
It makes zeroing z axis and sincronizing it with needle position much
easier.
Maybe it sounds complicated but it is actually very simple and elegant
solution, if you exactly know what you want to accomplish and what tools or
rules you have. Study how sewing machine works, how grbl and bcnc work and
how driller plugin works.

Good luck!

18.03.2019. 08.35 PM, "Frank Wilder" <notifications@github.com> је
написао/ла:

wilderf353, 2019-03-19:
I have been thinking about going at it differently before we started playing with bCNC.

I was thinking about putting a reed switch (a switch that is triggered by a magnet) near the handwheel and hooking it up to the z limit switch.  I would place a small magnet on the handwheel to trigger the reed switch when the wheel is in the position where the needle is at its highest point.  To control the sewing machine motor, I was planning to use the spindle drive connection with a relay.

I am a Python developer. I was thinking of writing a small g-code sender program. It would first send a gcode to turn on the motor and stop it when the reed switch closes (needle is up). It would then send g-codes normally (X & Y movements) until it got to some code to actually make a stitch (i.e. a Z call. ) It would then make a stitch (turn on spindle and wait for z limit switch trigger after 1 full rotation).  Once I get the z limit switch trigger, I would then send a gcode to turn off the spindle and then start sending g-codes again (X & Y movements).

I need to find a program/code to convert a bitmap to a gcode file.  I am thinking the file will be a series of (1) move to XY pos, and (2) move to Z pos, but I am not sure. 

I will take a look at the driller plugin. I have not thought of adding a stepper motor to drive the sewing machine...wouldn't that have to be an expensive, heavy-duty stepper motor?  

To make it work with bCNC, maybe I could add a second Arduino and put it between the Z stepper motor driver and the relay?  You could then move the reed switch connection from the CNC shield to this new Arduino and write some code for Arduino to make the stitch : (1) wait for the pin connected to the Z motor shield to go high, (2) power up relay to turn sewing machine motor (3) wait for pin connected to the reed switch to go high (3) turn off relay. 

Could I also use some pins on the Arduino to connect to Z limit switches on the stepper motor shield? I could use them to tell bCNC that the z motor is back it's "safe for movement" position.

FatNarval, 2019-03-19:
Yes that, sounds plausable, and for your skillset it should be piece of
cake.
My approach mostly uses existing software.

Nema 23 should do it,  maybe nema 17 too, but it also depends on particular
sewing machine .

When speaking about bcnc in context of embroidery, there are two planes of
the story: one is arduino grbl interface, and another is a driller plugin.

There are plenty grbl interface programs around there.

But the beauty is in Driller plugin which offers generating sewing or
embroidery Gcode where every stitch is predetermined.
And it seems that it could work without any real coding intervention which
is great for nonprogramming crowd.

the simplest/laziest solution is to ignore needle position completely
during feed moves, it is fabric after all,.
Only to hold needle in upper position only for rapid moves.

Problem for non stepper motor solutions is overshooting, solution are some
industrial sewing motors like servomotors or efka quickstop with
electricaly/electronicaly actuated clutch and brake or  drills-
screwdriwers with brakes.

In my oppinion,that is why using steper( or servo) controlled by z axis  as
a drive motor is elegant solution.

Harvie, 2022-04-07:
> just need to enhance the GUI sender so it can detect a broken thread 

If i remember correctly you can pause the bCNC sender by signaling proper GPIO input on GRBL, so you should wire your broken thread detector to that.

> handle color changes

This should be exactly same as toolchange that is already implemented in grbl/bcnc

So both of these probably don't need any modifications to be done at all

Ams-Laser, 2022-04-07:
just need to figure out a way to properly resume after a thread break, we need to have a resume function that will:
1 home Z (to lift the needle)
2 whait for either a restart signal or a step back signal
3 if step back is pressed we need to go back on gcode ignoring the Z moves, but keeping the current Z position value
4 on resume signal, run the resume stitch code (do some stitches with minimal xy moves so thread fixes to cloth) 
   then set Z to previous value by a G92Zxxx command then continue with gcode

hope this is of interest to the comunity

Edward