vlachoudis/bCNC #1047: Pocket Operation not working

micksulley, 2018-11-17:
This may be my lack of understanding, but I cannot get it to work.
I want to cut some stencil type text.  I have created it in LibreCAD and imported into dxf2gcode to generate the gcode, here is 0 as an example
[image]
As you can see there are connections top and bottom so that the centre does not drop out.
When I load the gcode into bCNC it works, but uses the lines as the centre of the cuts, so the connecting sections are almost gone.  The are options in dxf2gcode for left or right cutter compensation, but these use G41 and G42 commands which bCNC does not like.
I then thought I would use the bCNC tools pocket operation, so I have tried -
 - set the endmill to 3mm, profile direction to inside and check pocket option, click Pocket and run.  Cut exactly the same as before.  I did try outside as well but made no difference.
 - set profile to additional 4mm offset, click Pocket and run.  Again exactly the same as before.
Am I missing something here?  I couldn't find any documentation for this, is there any?
Thanks
Mick

MARIOBASZ, 2018-11-17:
Are you sure it takes more than one pass to clean all the inside of the pocket with that tool?

MARIOBASZ, 2018-11-17:
compensation 0, the profile route in in, or the outermost line in pocket operation) must be at 1.5 of the shapes. It does not happen?

micksulley, 2018-11-17:
I'm sorry but I don't understand your comment.  
I have run it with 'Additional offset distance' at zero and with it at 1.5 with Direction = inside and 1.5 with Direction = outside.  All 3 setting give exactly the same cut.  I assume that I am doing something wrong, but have no idea what.

MARIOBASZ, 2018-11-17:
Rare. Can you send the example to run and see if I understand the problem?

micksulley, 2018-11-17:
Here is the gcode as a zip
[test2noComp.gcode.zip](https://github.com/vlachoudis/bCNC/files/2592346/test2noComp.gcode.zip)

MARIOBASZ, 2018-11-19:
It works well for me (actually, a problem appears in the pocket that is calculated within the profile operation if you select them together). Parameters: End mil 3.00mm, stepover 40%
Observation: If overcut 4.00 mm, it does not generate a route: it is logical, there is no space inside.
Displays route difference with 0.3mm overcut, overcut -2mm

Harvie, 2018-11-19:
Is this the same case as #1046 ?